OpenEVSE/ESP8266_WiFi_v2.x #3: Whitespace is kind of a disaster ;)

sandeen, 2017-04-30:
What is the desired whitespace/indentation for the project?  The mix of tabs & spaces is kind of a disaster for readability; is there a desired editor setting to make it all come out right, or would a flag-day whitespace fixup be reasonable?

Tree-wide whitespace fixes can be nasty because patches won't apply well across the change, but a once and for all cleanup might help with code maintenance going forward ... Just a thought.

So in short: What is the expected coding style, and would you accept a patch to make it consistent?

Thanks,
-Eric

glynhudson, 2017-05-01:
Agree, consistency could be improved. Yes, we would accept a batch to fix consistency. No preference over tabs or spaces, happy to go with your judgment.

jeremypoulter, 2017-05-02:
I vote for Spaces, TABs are bad and inconsistent on different editors, etc. a space is a space no matter what the editor. Not fussed if 2 or 4, but would not go for more. 

Once we come to a conclusion I would suggest adding a [.editorconfig](http://editorconfig.org/) file to help people get the editors set up correctly.

sandeen, 2017-05-04:
Ok, I opened PR #9 w/ whitespace fixups.  I certainly don't want to offend anyone w/ my taste or style, so if it makes anybody's eyes hurt or twitch too much let me know, easy enough to run with a different set of indent rules.  :)  Take a look & see what you think.

sandeen, 2017-05-04:
PR #9 fixed this, I guess.

closed 2017-05-04 by sandeen