OpenEVSE/ESP8266_WiFi_v2.x #7: Can't update SPIFFS OTA

sandeen, 2017-05-02:
I know this is a known issue, but I'm wondering where things are at development-wise.  Chris had mentioned the inability to update SPIFFS over the air.

There seem to be 2 OTA update libraries; ESP8266HTTPUpdateServer and ESP8266httpUpdate.

I see that https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266httpUpdate/examples/httpUpdateSPIFFS allows a SPIFFs update, but it expects a URL to the update image, rather than a local file.  Is local file upload a requirement?

Thanks,
-Eric

sandeen, 2017-05-02:
Looking at the ESP8266httpUpdate library, I'm pretty sure the ESP8266HTTPUpdateServer library currently in use can be modified to update SPIFFS as well.  Just trying to find my way around the code, but having 2 different input forms, one for SPIFFS and one for firmware, doing appropriate size checks, and then calling Update.begin() with U_FLASH vs. U_SPIFFS depending on the form name seems like it would work...

sandeen, 2017-05-03:
Ok, @chris1howell this patch to ESP8266HTTPUpdateServer actually allows me to update SPIFFS through the same webform as you have for firmware.  May need a few more checks, but it works for me in simple testing.
I'm not quite sure how one stuffs a replacement for an arduino library into a project... 

```
--- /Users/sandeen/src/Arduino/libraries/ESP8266HTTPUpdateServer/src/ESP8266HTTPUpdateServer.cpp	2017-04-30 14:11:56.000000000 -0500
+++ ESP8266HTTPUpdateServer/src/ESP8266HTTPUpdateServer.cpp	2017-05-02 23:39:42.000000000 -0500
@@ -5,11 +5,20 @@
 #include <WiFiUdp.h>
 #include "ESP8266HTTPUpdateServer.h"
 
+extern "C" uint32_t _SPIFFS_start;
+extern "C" uint32_t _SPIFFS_end;
 
 const char* ESP8266HTTPUpdateServer::_serverIndex =
-R"(<html><body><form method='POST' action='' enctype='multipart/form-data'>
-                  <input type='file' name='update'>
-                  <input type='submit' value='Update'>
+R"(<html><body>
+<form method='POST' action='' enctype='multipart/form-data'>
+Firmware:
+                  <input type='file' name='firmware'>
+                  <input type='submit' name='firmware' value='Update'>
+               </form>
+<form method='POST' action='' enctype='multipart/form-data'>
+Spiffs:
+                  <input type='file' name='spiffs'>
+                  <input type='submit' name='spiffs' value='Update'>
                </form>
          </body></html>)";
 const char* ESP8266HTTPUpdateServer::_failedResponse = R"(Update Failed!)";
@@ -42,7 +51,9 @@
       if(!_authenticated)
         return _server->requestAuthentication();
       _server->send(200, "text/html", Update.hasError() ? _failedResponse : _successResponse);
-      ESP.restart();
+      if (_server->hasArg("firmware") && !Update.hasError()) {
+        ESP.restart();
+      }
     },[&](){
       // handler for the file upload, get's the sketch bytes, and writes
       // them through the Update object
@@ -61,9 +72,16 @@
         WiFiUDP::stopAll();
         if (_serial_output)
           Serial.printf("Update: %s\n", upload.filename.c_str());
-        uint32_t maxSketchSpace = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
-        if(!Update.begin(maxSketchSpace)){//start with max available size
-          if (_serial_output) Update.printError(Serial);
+        if (upload.name == "spiffs") {
+          size_t spiffsSize = ((size_t) &_SPIFFS_end - (size_t) &_SPIFFS_start);
+          if(!Update.begin(spiffsSize, U_SPIFFS)){//start with max available size
+            if (_serial_output) Update.printError(Serial);
+          }
+        } else { // firmware"
+          uint32_t maxSketchSpace = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;
+          if(!Update.begin(maxSketchSpace)){//start with max available size
+            if (_serial_output) Update.printError(Serial);
+          }
         }
       } else if(_authenticated && upload.status == UPLOAD_FILE_WRITE){
         if (_serial_output) Serial.printf(".");

```

glynhudson, 2017-05-03:
Nice work, it should be easy using platformIO. Just put the library in https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/tree/master/lib. 

@jeremypoulter do you know how to force platformIO to use a custom local lib rather than those included in the Arduino espressif8266 platform? 

What changes did you make to the library? Did you write the parch yourself? Can you get them merged into: https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266HTTPUpdateServer

jeremypoulter, 2017-05-03:
@glynhudson I think you are right putting the library in the lib directory should work. 

Alternatively you can fork the ESP8266 Arduino core and use that instead of the official one. I did that for one of my projects: https://github.com/jeremypoulter/SmartPlug

sandeen, 2017-05-03:
@glynhudson , yes I will see about getting it merged, it'll need a little bit of cleanup first, I think.  Probably needs to be configurable to choose between firmware vs. spiffs vs. both, for example.  As for changes I had to make, they're all in the patch I posted.  Essentially just need to detect whether the user has requested a flash update or a spiffs update, then call `Update.begin(spiffsSize, U_SPIFFS)` with the proper size and command (vs. [Update.begin(maxSketchSpace, U_FLASH)](url)).
One thing to note about the other updater (the one that works from a URL) is that it looks a bit more robust, because if I read it right, the server can send an md5sum of the image and it'll detect corruption.
Anyway, the local file spiffs updater is probably worth merging upstream in any case.

glynhudson, 2017-05-03:
Sounds good. It looks like there is already a PR open on Arduino code for SPIFFS update https://github.com/esp8266/Arduino/pull/2701

sandeen, 2017-05-03:
Eek, that one's a little hacky - `<input type='hidden' name='cmd' value='100'>` where `#define U_SPIFFS  100`.  And it doesn't check for spiffs space, etc.  Maybe if I get mine cleaned up I'll have more luck.

jeremypoulter, 2017-05-03:
@sandeen, agree, could be better ;-) esp8266/Arduino#2701 looks to essentially be dead so you may as well carry on and raise a PR. WTR your patch, you have two form upload boxes so you should allow for both the firmware and fs to be updated at the same time, or I would go for a <select> to specify which area to update.

sandeen, 2017-05-03:
@jeremypoulter yeah need to think a little bit about how these updates should work - firmware may depend on content and vice versa.  Not quite sure how (or if) any such dependencies should be enforced via the update tool...
I'm not sure the updater core will easily allow both to be updated at the same time.

sandeen, 2017-05-03:
FWIW I put a branch up with a local copy of the spiffs-capable updater at https://github.com/sandeen/ESP8266_WiFi_v2.x/tree/spiffs-demo if you want to play with it.
Ok now have to do some real work for the day :D

sandeen, 2017-05-03:
Hah, well, it worked on my sparkfun thing, so got brave and put it on my EVSE and it seems to crash it when I try.  Getting another huzzah in the mail, will get it figured out.

jeremypoulter, 2017-05-13:
FYI: This should be possible to implement directly in web_server.cpp once #16 is merged, this uses the Update class directly.

sandeen, 2017-05-13:
Oh, cool.  FWIW, I did open another PR for the ESP8266HTTPUpdateServer... hopefully it'll be useful to somebody.  :)  But yes, this looks cleaner than including the frame as it's done today, in any case.

sandeen, 2017-05-13:
(I'm kind of surprised nobody has defined a firmware+spiffs format with a header pointing to the different regions, to allow updating both at once - I bet we could actually hand-roll something to make that work, though)
`[[MAGIC][FW offset, length][SPIFFS offset, length]][<-- firmware -->][<-- spiffs -->]`
or something like that.

jeremypoulter, 2017-05-13:
Yeah, probably a lot of politics involved and lots of support/integration work ;-)

glynhudson, 2017-08-16:
Implemented in https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/pull/73 nice work @jeremypoulter! 👍

commit 320f4688 (2017-08-13): Updates to support SPIFFS update via HTTP upload

Resolves #7
closed 2017-08-16 by glynhudson