OpenEVSE/ESP8266_WiFi_v2.x #13: Several EVSE settings are read only once at bootup

sandeen, 2017-05-09:
There are several OpenEVSE values that may be changed at runtime but are polled only once, when the wifi boots up.[1]  On the other hand, they change infrequently enough that they probably don't need to be part of the primary RAPI polling loop.  Perhaps they should be directly updated if any of the input methods (RAPI screen, future config screen) are used to change their values.
[1] ammeter scale, charge limits, pilot value, safety check status

jeremypoulter, 2017-07-16:
FYI: This will probably be resolved as part of #34. This is most likely going to read values directly from the OpenEVSE via the RAPI HTTP endpoint.

jeremypoulter, 2017-07-30:
A lot of the RAPI values are now read as needed by the HTML/JS, see #51

closed 2017-07-31 by glynhudson