OpenEVSE/ESP8266_WiFi_v2.x #15: WiFi should be used to keep correct time

sandeen, 2017-05-09:
It would be very good if the network connectivity in the WiFi unit could be used to keep correct time in the EVSE.  There are a few options that I can think of:
- Full-blown [NTP client](http://playground.arduino.cc/Code/NTPclient)
- [Clever hack](http://playground.arduino.cc//Code/Webclient) to get time from a 3rd party webserver
- 3rd party Web API like that provided by [TimezoneDB](https://timezonedb.com/api)

The first two options have some degree of difficulty selecting and managing the many changing time zones, especially with daylight savings time changes.  (Both return time in GMT, and must be converted to local time).  The third option makes this quite easy (those smarts are on the other end), but it would likely require the user to sign up for another account to get an API key for the service.

We may also want to consider whether EVSE time should be kept in GMT, or in local time.

Another small wrinkle would be how to handle an active scheduled charging session and/or time-limited session if the clock changes underneath it.

jeremypoulter, 2017-05-28:
One other option would be to use Javascript to pull the time from the PC when on the Web UI.

Personally I think this + NTP would be sufficient.

sandeen, 2017-05-28:
My motivation for this was handling daylight savings changes, as some people on this menof day plans were burned by that.  I'm not sure counting on a client connection to fix it would suffice.  And connecting from another time zone could cause issues....  NTP is simple enough but handling daylight savings gets a bit tricky I think.

sandeen, 2017-05-30:
I do have some test code that uses the free [timezonedb.com](timezonedb.com) API to query time and update the clock once per hour (sooner if a DST change is pending) .  If anyone is interested let me know and I can clean it up and push out a branch.

sandeen, 2017-05-30:
Using NTP and managing DST on our own would require some mechanism to correlate the user's selected time zone to an entry in the time zone databasein order to anticipate the next change, and we wouldn't want to keep all that TZ data around in the image.  It's quite large and almost all of it is not relevant to the user's time zone.
We could alternately support only a handful of timezones in the firmware, but that doesn't seem great either.  By now users are used to creating accounts and getting API keys, so using the timezonedb service may not be so bad ...

jeremypoulter, 2017-07-24:
I have added setting the time to the same as the client time in #34, works quite well for the basic use case of just setting the time.

glynhudson, 2017-07-27:
Can confirm, time setting as introduced by #51 works well :+1:

closed 2017-07-27 by glynhudson