OpenEVSE/ESP8266_WiFi_v2.x #28: Recent changes seem to have mismerged

sandeen, 2017-05-27:
After commit f7c61ca things seem to have gone off the rails a bit - perhaps a mis-merge, I can't tell for sure, some of the changes are dated March if I use git blame...

- `#define DEBUG_SERIAL1` got commented out in src/emonesp.h, and now debug messages are going out on the main serial line.
- The solar divert tab is also now visible by default (was that intentional?)
- home.html got renamed to home.htm
- input.cpp gained new whitespace damage
- etc.

Did something go wrong on recent commits to master?

sandeen, 2017-05-27:
There are also new EEPROM locations inserted into the middle of the list, which makes any older values stored come out in the wrong slots with newer firmware.  This will be a problem for any  in-production upgrades.

jeremypoulter, 2017-05-27:
While I agree the values shouldn't have been added in the middle, an
unfortunate side effect of adding the checksum is that it invalidates all
the config values anyway (and I am not sure there is a way around it) so
probably ok in this instance.

I will raise an issue about changing the config to use JSON.

sandeen, 2017-05-27:
Ok interesting point about the checksums.
Is an upgrade / conversion path between versions not a requirement?  As more services get added that sounds like a bit of a hassle for users.

jeremypoulter, 2017-05-28:
Now the config has a checksum we can know if the config is valid or not, so hopefully any future updates can be gracefully migrated, however I don't see any clean way of migrating the old config... hum actually there might be a way... the strings are followed by 0s where as an erased flash sector will be 0xff so maybe this can be used. I will move the new values to the end and write some simple code to detect what looks like a valid old config value and migrate.

sandeen, 2017-05-28:
Cool, thanks - although it crossed my mind that if Chris doesn't make a binary release before you switch to JSON, that's the only transition that will matter in the real world ...

sandeen, 2017-05-30:
I don't know what's going on in the tree at this point, I'm also failing to load the home.htm[l] unless I hit reload many many times.  It loads a couple of resources but hangs on others.  Sorry for the slew of nonspecific bug reports.

jeremypoulter, 2017-05-30:
Can you try #26 and see if that is reporting any slow parts of the system?

jeremypoulter, 2017-05-30:
Also do you have MQTT and EmonCMS setup?

sandeen, 2017-05-30:
I don't have them set up.  This was  fresh update after zeroing the flash.  Took a few tries to get through the AP selection, got that done, then the normal homepage via the client is unhappy.
[image]
shift-reload:
[image]
cancel:
[image]
The ESP is still alive, as evidenced by the RAPI commands  continuing to come out on the console.

sandeen, 2017-05-30:
@jeremypoulter so is the master branch in the tree currently as intended?  Sorry for harping on that but when I see things like
0e75019 Merge branch 'master' of https://github.com/jeremypoulter/ESP8266_WiFi_v2.x **into solarpv**
with lots of seemingly disparate changes that actually hit the **master** branch:
```
 data/config.js |   19 -
 data/home.htm  |   38 ---
 input.cpp      |  155 +++++++-------
 input.h        |   14 -
 src.ino        |    7 
 web_server.cpp |  599 ++++++++++++++++++++++++++++++++++-----------------------
 6 files changed, 462 insertions(+), 370 deletions(-)
```
I'm confused.  It seems like changes snuck in that weren't intended, but if I'm wrong, please set me straight.  Did you & @glynhudson intend to enable the solar divert code in the master branch?
I'm sure we can whack out the bugs one at a time, I'm just trying to figure out what happened in the tree.
Thanks,
-Eric

jeremypoulter, 2017-05-30:
Yes the solar branch got merged into the master, I am not entirely sure if that was intended or not. 

@glynhudson I am thinking it may be best to roll back those changes, ideally just reset the master back to fdaeb1319493977c832f630a7ea4ac5a2c121cbf and do a force push. Might be a pain for us to update but will end up cleaner in the end when merging back the solar PV branch.

jeremypoulter, 2017-05-30:
Actually totally ignore that the solarpv was not the last commit, going to either have to revert the Solar PV changes, or just live with it and fix the issues.

sandeen, 2017-05-30:
Depending on what happened, I could live with a rebase & re-assembly of the subsequent patches, FWIW - if that makes sense.

Things are OK for me as of f7c61ca at least.

sandeen, 2017-05-30:
Nothing is spitting out from me on the profiling, FWIW.  (meaning nothing is slower than 10 millis I guess?)

jeremypoulter, 2017-05-30:
Hum... odd is it enabled? you need -DENABLE_PROFILE

glynhudson, 2017-06-01:
Sorry for the lack of communication. I've been away last week and struggling to catch up.  

Oh dear merging solarpv into master was not intended. It sees when I merged PR #17 I much have been on the solarpv branch therefore accidentally merged the solar PV into master 😞 .  Mmm how shall we proceed? I think it's best to keep the changes in master, solarpv was almost ready to merge anyway. I'll do some testing this week. 

Sorry commenting out `#define DEBUG_SERIAL` was my doing, I was just testing and forgot to revert before pushing. Is there a clever way to define `DEBUG_SERIAL` in a platformio env therefore it can easily be turned on for testing? Will the `-DENABLE_DEBUG` flag in the `env:openevse_ota` platformio environment enable serial debug? 

Renaming home .html > .html was also me. This was done a while back when I was struggling with ESP web server performance issues. I read that apparently the old ESP web server worked better with .htm rather than html. It's not an issue with async web server. Do you think we should rename back to html? I don't think it makes a difference,.

sandeen, 2017-06-01:
I don't think htm vs html really matters, it just seemed further indicative of unintentional changes.
I've got weird stuff going on and can't tell why, it seems like a lot of changes got lumped in and it's tough to bisect.

jeremypoulter, 2017-06-05:
Ok, I think we have resolved what went wrong, so I would suggest closing this issue and raising a separate bug report for any remaining issues.

sandeen, 2017-06-05:
Sounds fine - I have been gone and haven't had a chance to do any testing recently, FWIW.

closed 2017-06-05 by sandeen