OpenEVSE/ESP8266_WiFi_v2.x #34: Replicate OpenEVSE local (LCD) controls on web UI

jeremypoulter, 2017-05-28:
All the controls on the OpenEVSE LCD should be available on the web UI with a focus on;

- [ ] Change charge current limit
- [ ] Set the time limit
- [ ] Setting the time, see #15 
- [ ] Set the service Level
    - Use location to set appropriately for the region?

glynhudson, 2017-05-31:
Yes, service level will always be L2 in countries that have 240V. In The USA is could be L1 or L2 depending on the outlet it's plugged into. However, I guessing most (if not all) users will be using L2 from 240V outlets since much of the benefit of using a unit like the OpeneEVSE if it's connected  110V  outlet. What do you think @chris1howell

chris1howell, 2017-05-31:
Most of the world will be L2 only, in the US and other countries that use 120 and 240v L2 will be used most often. Many users in the US build the kit that can support both and OpenEVSE will be set to auto. The most common settings are L2 only or Auto. L1 only is rarely used.

I do not think there is much benefit to set by region, it is best to just use the default from firmware and allow the user to change if necessary.

jeremypoulter, 2017-07-26:
All changes now implemented in #51

closed 2017-08-03 by jeremypoulter