OpenEVSE/ESP8266_WiFi_v2.x #36: Change config to be stored as JSON

jeremypoulter, 2017-05-28:
The current method for saving the config is not very flexible or future proof. As the ESP8266 has 4K of just saving a JSON block would allow for a bit more flexibility, eg https://github.com/jeremypoulter/SmartPlug/blob/master/src/config.cpp

omersiar, 2017-09-06:
In my project i am storing config file on SPIFFS as JSON object. ArduinoJson library makes all easy. 

https://github.com/omersiar/esp-rfid/blob/7fd15ea5a290c531cbd88b7d2e366ae4392a6105/esp-rfid.ino#L407-L417

Also two-way WebSocket communication between ESP and Browser simplifies most of the things.

You can use new function for network surveying and send a JSON object to browser when it is ready.

`WiFi.scanNetworksAsync(printScanResult, true);`

```arduino
// Send Scanned SSIDs to websocket clients as JSON object
void printScanResult(int networksFound) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["command"] = "ssidlist";
  JsonArray& scan = root.createNestedArray("list");
  for (int i = 0; i < networksFound; ++i) {
    JsonObject& item = scan.createNestedObject();
    // Print SSID for each network found
    item["ssid"] = WiFi.SSID(i);
    item["bssid"] = WiFi.BSSIDstr(i);
    item["rssi"] = WiFi.RSSI(i);
    item["channel"] = WiFi.channel(i);
    item["enctype"] = WiFi.encryptionType(i);
    item["hidden"] = WiFi.isHidden(i)?true:false;
  }
  size_t len = root.measureLength();
  AsyncWebSocketMessageBuffer * buffer = ws.makeBuffer(len); //  creates a buffer (len + 1) for you.
  if (buffer) {
    root.printTo((char *)buffer->get(), len + 1);
    ws.textAll(buffer);
  }
  WiFi.scanDelete();
}
```

Check it out we can share many things. https://github.com/omersiar/esp-rfid

glynhudson, 2020-07-28:
Now implemented

closed 2020-07-28 by glynhudson