OpenEVSE/ESP8266_WiFi_v2.x #41: Web page connection hanging at first load

glynhudson, 2017-06-12:
I've just been testing the latest changes and I find that the webpage seems to hang on "loading please wait" when I try and load the web page for the first time via the captive portal. 

If i revert back to c95ef7d this seems to work fine, then moving forward a few commits to e6b65c3 the hanging issue returns. It seems it's just the web server that's hanging. The serial port shows the device is still issuing RAPI commands. 

Do you think the increase in `EEPROM_SIZE`  increase to `4096` might be causing an issue? I'm testing on Adafruit Huzzah. 

I will debug further tomorrow. Just wanted to quickly confirm with @jeremypoulter that the latest master branch is working fine for you? Maybe it's just me!

jeremypoulter, 2017-06-12:
Indeed, I will take a look.

roy13, 2017-06-15:
I'm also getting the hanging on the captive portal.

jeremypoulter, 2017-06-15:
I have been looking in to this, I think the https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/c95ef7da5d6509d02513a94cce41b6da00d9bcf1 was a bit of a red herring, I was able to reproduce the issue in that build.

As far as I can see there is an issue with serving static files in AP mode. I am trying to trace it down.

jeremypoulter, 2017-06-15:
So I think the issue is down to running out of memory. So in answer to @glynhudson the increase of the EEPROM_SIZE to is partly to blame but also just a general increase in memory usage. The reason it is happening in AP mode is down to running the AP needing more memory.

glynhudson, 2017-06-17:
Ah good effort tracking this down. How did you go about figuring out that it was a memory issue? Is there a way to measure free memory? What a pain. What are the options available to us? Will we have to cut down on some features to optimise?

jeremypoulter, 2017-06-19:
Found it by just basically printing out the free memory and watching it. I am going to review the code both ESP8266_WiFi_v2.x and ESPAsyncWebServer and see if there are any quick wins.

glynhudson, 2017-06-29:
For anyone looking at this issue, I have done a stable release prior to the introduction of this bug. Please use the stable release (2.1.0) rather than latest master for the moment: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/releases/tag/2.1.0

glynhudson, 2017-07-01:
While we're working on this I think it would be a good idea to revert the master branch to [stable release 2.1.0](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/releases/tag/2.1.0) at https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/9838d38a99a77a53074af89ed9f55f771b15ab5b and move the curren master to a 'dev' branch. I don't want to inconvenience any users who might try and update to latest master. 

Would you be happy to do this or shall I?

chris1howell, 2017-07-01:
Yeah good idea.

Would you mind doing it? I am not as proficient with github as I should be.

jeremypoulter, 2017-07-01:
Agree, probably a good idea, although it may be easier to create a stable
branch from release 2.1.0<https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/
releases/tag/2.1.0> at 9838d38<https://github.com/
OpenEVSE/ESP8266_WiFi_v2.x/commit/9838d38a99a77a53074af89ed9f55f771b15ab5b>
rather than roll back/revert master.

Jeremy

OpenEVSE/ESP8266_WiFi_v2.x/releases/tag/2.1.0> at 9838d38<
https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/
9838d38a99a77a53074af89ed9f55f771b15ab5b> and move the curren master to a

OpenEVSE/ESP8266_WiFi_v2.x/issues/41#issuecomment-312400411>, or mute the
ABvYH6VuzUxMMH8WG7gbZqCgLzdb73xkks5sJZfFgaJpZM4N3VPB>.

glynhudson, 2017-07-07:
Good idea, I have made a `stable` branch with 2.1.0 release. I have made the `stable` branch the default branch: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/tree/stable

I have removed the `dev` branch to avoid confusion. The `master` branch shall be used as the dev branch

glynhudson, 2017-07-27:
This issue has now largely been fixed by #50 and #46.

closed 2017-07-27 by glynhudson