OpenEVSE/ESP8266_WiFi_v2.x #49: Add support for setting the WiFi mode from the charging station

jeremypoulter, 2017-07-18:
See https://github.com/lincomatic/open_evse/blob/stable/rapi_proc.h#L52

jeremypoulter, 2017-07-27:
Rather than switching to AP mode it may be handy to use the long button press to enable [WPS](https://github.com/arendst/Sonoff-Tasmota/blob/master/sonoff/support.ino#L278:L325).

jeremypoulter, 2018-01-13:
This is resolved now

closed 2018-01-13 by jeremypoulter

glynhudson, 2018-02-01:
Just a note for the future, I realise the ESP wifi gateway does not utilise the open_evse controller feature where a long 10s press of the menu button will result in `WIFI_MODE_AP_DEFAULT` being printed to serial. 

Maybe this could be used a fail save factory reset of ESP wifi-gateway settings? 

https://github.com/lincomatic/open_evse/blob/development/firmware/open_evse/rapi_proc.h#L72

```
$WF mode\r - Request client WiFi mode
 mode: WIFI_MODE_XXX
 (currently very long press (10 sec) of menu btn on OpenEVSE will send WIFI_MODE_AP_DEFAULT
v2.0.1+: 2-hex-digit XOR checksum appended to asynchronous messages
```