OpenEVSE/ESP8266_WiFi_v2.x #52: Solar PV divert (to do)

glynhudson, 2017-07-27:
To summarise my discussion with @jeremypoulter this afternoon. Solar PV divert should:
- [ ] Only adjust charging current up to the maximum already set in the OpenEVSE
- [ ] When diversion charging has fished (in Eco mode) or car is unplugged the Openevse should revert back to normal mode ready for the next charge. By default the car should always start a normal charge when it's plugged in
- [ ] The RAPI commands sent by the divert function should be changed to use the rapi sender class which will add the checksum and be more robust. 
- [ ] The text information in the html page should be changed to info pop out boxes 
- [ ] The Solar PV divert buttons (eco / normal) should be moved to the main `OpenEVSE`tab and the `Mode` button should be removed 
- [ ] Solar PV diversion to be tested using MQTT nodeRED solar PV simulator: https://github.com/openenergymonitor/openevse-solarpv-divert-nodered

jeremypoulter, 2017-08-03:
> The text information in the html page should be changed to info pop out boxes
> The Solar PV divert buttons (eco / normal) should be moved to the main OpenEVSEtab and the Mode button should be removed

Done in #58

jeremypoulter, 2017-08-08:
@chris1howell @glynhudson @lincomatic 

With regard to the remaining item;
> When diversion charging has finished (in Eco mode) or car is unplugged the Openevse should revert back to normal mode ready for the next charge. By default the car should always start a normal charge when it's plugged in

I assume the best way to detect charging finishing is the state changing from 3 to another state? Or should I also watch something else as well? amps?

glynhudson, 2017-08-10:
The car being unplugged status is the best way to detect charging has finished IMO. The car will usually stop charging (since it's battery is full) before being unplugged, however there would be no reason to revert back to normal (max power) charging mode until EV is un-plugged. I think this would also be the simplest technically to implement?

closed 2017-09-05 by glynhudson