OpenEVSE/ESP8266_WiFi_v2.x #53: RAPI over MQTT broken?

glynhudson, 2017-07-27:
Just tested RAPI over MQTT which was working reliably in the past. Now seems not to work running latest master with PR #51 

Publishing a RAPI command to MQTT e.g. `openevse/rapi/in $GC` (assuming `openevse` is the base MQTT topic set on the ESP) should result in a RAPI response being posting to `openevse/rapi/out`. There is no response. 

Looking at the debug output I don't think the MQTT message is getting received since there is no debug output to indicate. "MQTT received" should appear in the debug ouput when an MQTT message is received: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/blob/stable/src/mqtt.cpp#L33

MQTT status is publishing nicely has has been reliable for 20hr :+1:

glynhudson, 2017-07-29:
Update: After further testing I've established that the rapid MQTT is being received by the unit e.g. 

publishing `openevse/rapi/in/$GC` results in a serial debug:

```
MQTT received:
openevse/rapi/in/$GC 
```
However, the rapid command is not acted upon, no output is published to `openevse/rapi/out`

jeremypoulter, 2017-07-30:
Looks like MQTT was never changed to use the RAPI Sender, should probably update to use that rather than fixing the existing code.

PR #54 (2017-07-31): Fix for RAPI via MQTT
The Solar divert logic was using = rather than == when checking the topic, fixes #53.

Also changed to use RapiSender.
closed 2017-08-02 by glynhudson