OpenEVSE/ESP8266_WiFi_v2.x #55: Turn off AP / reboot after first connection to WiFi

glynhudson, 2017-07-31:
After connecting to a Wifi network, a reboot seems to be required before I can load the webpage. 

This sometimes happens because my laptop / mobile device is still connected to the openevse AP

Could a possible solution be to force an ESP reboot after connecting to Wifi for first time? This would have the added advantage of printing the new IP address of the ESP on the network to the LCD. 

Or maybe just turning off AP will fix the issue?

glynhudson, 2017-09-05:
Has this been implemented? I just tired to test by factory resetting the WiFi unit, however I accidentally entered the incorrect WiFi password when logged into the AP, I'm not locked out of the unit since the AP has switched off but the unit can't connect to the required SSID! I will have to clear the config flash via serial. Shall I open a new issue for this?

jeremypoulter, 2017-09-05:
Not been implemented yet I have started work on it, see https://github.com/jeremypoulter/ESP8266_WiFi_v2.x/tree/install_improvements, but very much WIP.

In the current firmware I believe it should start the AP if it does not connect to the WiFi in STA mode.

jeremypoulter, 2017-09-18:
Fixed after #87

closed 2017-09-18 by jeremypoulter