OpenEVSE/ESP8266_WiFi_v2.x #64: Publish state name string and session energy MQTT

glynhudson, 2017-08-05:
Instead of publishing state number (e.g. 1 means not connected) to MQTT it would be more useful to publish the string "Not Connected" to MQTT state topic. This would make integration with other home automation platforms much easier. e.g. the state string could be displayed in openHAB directly from MQTT without having to re-create a lookup. 

Likewise having session energy (Kwh) published to MQTT would also be very useful to integrate with other platforms. e.g. display how many Kwh the car has charged in this session in OpenHAB / Home assistant / amazon echo etc.

jeremypoulter, 2017-08-05:
I am not entirely sure I agree, while I don't like magic numbers, there are problems with text, mainly contex and translation, but also additional memory/processing requirements. Consider the changes/inconsistencies within just between OpenEVSE and OpenEVSE WiFi, Sleeping/Waiting, EV Contented/Connected, etc. 

I would recommend some kind of human readable token that you wouldn't directly display but is still a little for understandable, eg not_connected.

glynhudson, 2017-08-05:
Good point, memory processing requirements is an important issue. Let's stick with magic numbers for the moment, it's not that that to work with.  

There is also the issue that Emoncms does not support non-numerical input therefore the MQTT publishing section would have to be re-written independently to the Emoncms post. Currently exactly the same data is posted both to Emoncms and MQTT which I guess is good for consistency. 

Just enabled my Google Home to start / pause / enable eco mode on the OpeneEVSE  via voice using [ha-bridge](https://github.com/bwssytems/ha-bridge) RAPI MQTT, magic!👍 

Are you happy if I added session Kwh to MQTT and Emoncms?  This will also be useful for Emoncms to be able to create a proper OpenEVSE dashboard. It's more interesting to know how much charge is put into the car per session as well as the usual Emoncms day / week / month energy accumulations etc.

jeremypoulter, 2017-08-05:
Yeah the KWh is fine, no problem

chris1howell, 2017-08-05:
Yeah session kWh would be fantastic to send to EmonCMS.

jeremypoulter, 2017-08-11:
@glynhudson Did you do the KWh?

glynhudson, 2017-08-11:
> @glynhudson Did you do the KWh?

Yes, I added posting Wh to MQTT and Emoncms in: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/1257d258de023c2798a8f996bd3798c9cc061e02 👍 

Sorry, forgot to close this issue.

closed 2017-08-11 by glynhudson