OpenEVSE/ESP8266_WiFi_v2.x #65: Divert UI buttons don't persist state

glynhudson, 2017-08-06:
Just tested #62 , it looks like the solar PV divert buttons don't persist the mode. 

e.g. if I select "Eco Mode", eco mode gets correctly set (I can see via serial output).

[image]

Then if I refresh the web page the divert buttons go back to being unselected:

[image]

It's also not clear from the colours of the buttons which mode is selected. Currently white = selected and blue/green = not selected. I'm not sure how to improve this, maybe it is clear enough? Ideas welcome, I'm not an UI expert!

jeremypoulter, 2017-08-06:
Yeah, I noticed this, they also do not change if set via MQTT. I think this is down to overloading the divertmode to also handle is if the buttons can be set or not. 

I think it may be best to separate the logic of if the divertmode can be set or not from the actual mode.

Withe regard to the UI is was thinking maybe the toggle switch would be better than the buttons? On is Eco mode, off normal? Or are there more modes planned? In which case maybe a drop-down?

glynhudson, 2017-08-06:
Good idea, toggle would be much better than buttons. A toggle would be totally unambiguous. 

No new modes are planned currently, however I did think that in the future additional modes could be added. If this was the case it would be easy enough to change the UI to drop down in the future.

jeremypoulter, 2017-08-08:
Fixed in #68

closed 2017-08-08 by jeremypoulter