OpenEVSE/ESP8266_WiFi_v2.x #66: Simple Status UI: elapsed time?

glynhudson, 2017-08-07:
I've made a start at making a simple status UI box to sit at the top of the page. The idea is that this status UI should show the user at a glance the most important things. e.g is the car charging? At what rate? How many Kwh so far? 

One useful metric that we're not currently displaying is elapsed time. Elapsed time (in seconds) is returned in the 2nd element via RAPI get status `$GS`. @jeremypoulter do you think it would be possible to get elapsed time (in minutes) onto the UI display and add it to my status UI box as a third column? 

I think it would be also good if the status message e.g. "Charging / Waiting etc." was highlighted in the same color and the LCD. e.g. green = charging , purple = sleeping / waiting.

Any feedback on this proposed status UI box? Do you think it' useful? 

[image]

jeremypoulter, 2017-08-08:
How about;

[image]

[image]

pushed those changes to the branch

glynhudson, 2017-08-08:
That's great! Love it, exactly what I had in mind. 

Could the colour glow match the LCD background colours?

What do you think @chris1howell ?

jeremypoulter, 2017-08-08:
> Could the colour glow match the LCD background colours?

It does (kind of), the actual LCD colours look terrible in the UI, in this example the blue is close to the background of the LCD in charging mode (as opposed to cyan of the LCD text). The shadow is the same colour as the left border but with an opacity applied so will naturally be darker as with the LCD, but that being said I have no objections to tweaking the colours.

chris1howell, 2017-08-08:
Beautiful... Love it!

jeremypoulter, 2017-08-08:
FYI: not that these pictures really do the LCD justice.... 

[image]
[image]

[image]
[image]

[image]
[image]

glynhudson, 2017-08-09:
Awsome! Looks great. Nice work :+1:

jeremypoulter, 2017-08-09:
Just for reference some notes from Hangout discussion with @glynhudson.

> @glynhudson: thanks alot, I'll give it a try this evening. Instead of having a status only page I was thinking of having an 'Advanced' section at the bottom of the page which would show a few elements that would be hidden by default 

[image]

> @glynhudson: e.,g. the elements marked in red could be hidden by default then shown if a users taps an "advanced" button. Maybe this would be simpler than making more tabs? 

> @glynhudson: Would like to get to the stage where a user had everything they need for day-to-day use on a single tab. e.g. setting timers, charging status. 

> @jeremypoulter: Safety should probably not be hidden if any of the tests have failed or disabled

> @glynhudson: yes, if dev mode is not enabled 

> @jeremypoulter: I guess enabling dev mode should explicitly enable advanced mode

> @glynhudson: yes, why not

jeremypoulter, 2017-08-11:
Implemented in #71

closed 2017-08-11 by jeremypoulter