OpenEVSE/ESP8266_WiFi_v2.x #69: Charging timer is not reloaded in UI after page refresh

glynhudson, 2017-08-09:
If the charging timer (start / stop time) is set via the web UI or LCD then the web page is refreshed the UI does not display the current set start stop time.

jeremypoulter, 2017-08-09:
This is due to lincomatic/open_evse/#68. Can implement this now but only on the dev firmware.

glynhudson, 2017-08-09:
Ah yes, I remember seeing that issue. Would it be possible to implement it in a way that if users are running the dev FW then the current delay time state is displayed? I plan to ship all OpenEVSE units with the new FW. @chris1howell is working on test stable release. The dev FW has been working well for me so far so should all be fine.

jeremypoulter, 2017-08-09:
Yes I think that is possible, I think I can also do a [polyfill](https://en.wikipedia.org/wiki/Polyfill) of sorts on the ESP, eg in RAPISender intercept calls to the new API and on the older firmware remember the settings on the ESP. Won't survive a reboot of the ESP, but probably ok.

closed 2017-08-15 by jeremypoulter