OpenEVSE/ESP8266_WiFi_v2.x #70: Set HTTP Port?

glynhudson, 2017-08-09:
Currently port 80 is hard coded for the HTTP web server: 

https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/blob/master/src/web_server.cpp#L18

Is there any room left in the EEPROM to have the HTTP port a configurable option? This would be useful when running alongside other servers on the same domain. e.g. accessing openevse externally via dynamic DNS

alsavage, 2020-07-29:
tl;dr: I vote for this enhancement.

This would have been enormously helpful for me Nov2019.  I run a local httpd, so I needed to port forward some other external port to local-IP:80 in order to obtain outside access to the unit.  That, in turn, required a move to running dd-wrt on my router, as the router's built-in port forwarding capabilities would not allow disparate external<->internal port mappings.  It was rather more work than really necessary.

chris1howell, 2020-07-29:
It is pretty risky to expose unencrypted http to the internet.

Once 3.x supports HTTPS changing the default secure port 443 to another
port would not be a problem.

digitalarchitects, 2020-07-30:
Hi,
There are so many good ways that networking equipment and configs can resolve port and security issues that those should never impact design. Secure by default should be the aim, so HTTPS on any chosen port should be allowed.

Most container deployments map ports or include nginx to so do so and also hold a single certificate to encrypt all internal services.

regards,

Kevin Wilson

+27 11 552 4093 (Office)
+27 82 901 4224 (Mobile)
+27 86 676 6065 (DigiFax)

Stefanutti Stocks Corporate Services
a division of Stefanutti Stocks (Pty) Ltd
Reg No. 2003/022221/07

[cid:f169fe9a-0c8d-448d-b81e-27a13a9b1eef]

alsavage, 2020-07-30:
If I'm understanding Chris' comment correctly, https will be implemented in firmware 3.x (current is 2.9.1), and when that occurs the http/s port will become user-configurable.  My understanding is that https on ESP8266 is costly in terms of memory and CPU cycles.  Perhaps this is better suited for the ESP32, with its dual-core and faster base frequency?

digitalarchitects, 2020-08-03:
Hi Everyone,
I would suggest moving development to ESP32 as well, the limitations will just increase going forward. OTA upgrades are tricky if your binary gets too big, which leaves you with physical access for upgrades, not a great maintenance idea. The cost difference is negligible, $6 vs $3.

regards,

Kevin Wilson

+27 11 552 4093 (Office)
+27 82 901 4224 (Mobile)
+27 86 676 6065 (DigiFax)

Stefanutti Stocks Corporate Services
a division of Stefanutti Stocks (Pty) Ltd
Reg No. 2003/022221/07

[cid:e31f0313-186e-457e-8d47-643d958ba1aa]

jeremypoulter, 2020-08-03:
@digitalarchitects FYI we already have a port to the ESP32 https://github.com/OpenEVSE/ESP32_WiFi_V3.x.
@alsavage The ESP32 build will have support for the server to run over HTTPS, that is what Chris means by it will be supported in the 3.x version

digitalarchitects, 2020-08-04:
Is it a lot of work to support both? If so would it not be simpler to support only esp32? One code base with all the effort behind that.

If not, and there is good reason for both then I am happy to defer to your path.

regards,

Kevin Wilson

+27 11 552 4093 (Office)
+27 82 901 4224 (Mobile)
+27 86 676 6065 (DigiFax)

Stefanutti Stocks Corporate Services
a division of Stefanutti Stocks (Pty) Ltd
Reg No. 2003/022221/07

[cid:47c47b3d-a9bd-4ac8-b07e-c191feb80594]

glynhudson, 2020-08-04:
Only the ESP32 will be supported moving forward. It's likely that V2.9.1 will be the last update to the ESP8266

alsavage, 2020-08-05:
In another arena, Chris just confirmed:

_"We have been shipping the Huzzah ESP32 in all products for several months now. We are currently transitioning to a OpenEVSE designed custom ESP32 board with the power supply built in."_

Myself, I'm upgrading my ESP8266 board out of my system, and not agonizing about the feature I want not being brought to the older IC.

But I do want the subject feature (https, configurable port number) to be implemented someday ;)

glynhudson, 2020-08-05:
Correct, we (OpenEnergyMonitor) are also now shipping the ESP32 as standard with all European EVSE's https://shop.openenergymonitor.com/evse/

HTTPS is already implemented on the ESP32 https://github.com/OpenEVSE/ESP32_WiFi_V3.x

closed 2020-08-05 by glynhudson