OpenEVSE/ESP8266_WiFi_v2.x #72: Session Wh counter does not reset if using delay timer?

glynhudson, 2017-08-09:
In the past when charging with my OpenEVSE I've always used the onboard charging timer in my Nissan LEAF, however now I have the ability to easily set the OpenEVSE delay timer via the Wifi 2.0 GUI interface I have now turned off the car internal timer and am now using the OpenEVSE delay timer to schedule charging overnight. 

Once issue I've noticed is the OpenEVSE Wh session counter does not reset in between charging sessions when the delay timer is active. I.e previously when the car was unplugged then replugged in the OpenEVSE session wh counter would reset. 

@chris1howell  Is this a known bug in 3.11.3? Or is it due to the Wifi 2.0 FW?

glynhudson, 2017-08-16:
Known issue with OpenEVSE FW, confirmed by @chris1howell

glynhudson, 2017-11-14:
Issue opened on open_evse FW repo: https://github.com/lincomatic/open_evse/issues/76

closed 2017-11-14 by glynhudson

glynhudson, 2018-02-01:
Fixed in open_evse FW 4.10.4. Thanks @lincomatic 👍

glynhudson, 2018-10-03:
Hi @teslatradeup ,

You can download pre-compiled open_evse FW from here: https://github.com/openenergymonitor/open_evse/releases/

Please note that these are EU firmware releases, see these changes listed here: https://github.com/openenergymonitor/open_evse#changes-by-openenergymonitor

Be sure to download the OpenEVSE release not the EmonEVSE if you are using an OpenEVSE.

teslatradeup, 2018-10-03:
Oh man, I'm in Canada so I guess this EU firmware wouldnt work for me right?

Do you know if there is a US repo that has been updated by any chance?

Thanks for your help!

glynhudson, 2018-10-03:
I will work, the main changes are L2 by default and limit max charging
current to 32A. If you can live with these changes then it will work just
fine.

- sent from my mobile device

teslatradeup, 2018-10-03:
Ok perfect - that's actually fine by me. Thanks a lot! :)

Martin Provost

chris1howell, 2018-10-03:
Yes, we are testing a 5.0 release now. It should be uploaded in a couple days.

I would be interested in a US updated version as well, since I can charge
at 40AMPS and wouldn't want to lose that. Any possibility of that/or
updating the version on Chris's side?
https://github.com/OpenEVSE/open_evse/releases

Thanks!

cmgj74, 2018-10-03:
Awesome!  Thanks Chris!