OpenEVSE/ESP8266_WiFi_v2.x #75: Web page hanging, browser refresh required to fix

glynhudson, 2017-08-16:
Often when trying to load the web page from the ESP the page hangs on "loading", after a browser refresh (sometimes two is required) the page loads. Here is a network activity capture showing the page hanging the subsequently loading after a page refresh:

[image]

[image]

roy13, 2017-09-07:
This feels similar to the issue i raised back in #41. I've been content to continue with original WiFi RAPI but i'm really keen to try MQTT. Anyway i've flashed a brand new Node MCU 12E using Arduino and i get stuck on the first image above until the WiFi AP seems to drop out. No amount of refreshing seems to assist.

jeremypoulter, 2017-09-18:
Should be resolved after #87

closed 2017-09-18 by jeremypoulter