OpenEVSE/ESP8266_WiFi_v2.x #76: Auto set current time not working

glynhudson, 2017-08-16:
From this discussion https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/15 I remember that the auto time should be obtained from the web browser. I remember when I initially tested this when #51 was merged it seemed to work. 

However testing again now it seems not to work, see attached. Tested in both firefox and Chrome

[image]

glynhudson, 2017-08-16:
Ah, I just realised I needed to hit the "Set" button! :+1:  

Having the 'Automatic" toggle is slightly misleading as I assumed that having automatic enabled would update the time automatically. Is this possible? To update the time each time the page is opened? Maybe an overkill? However this would automatically handle daylight saving time updates, since the web browser time would automatically update then the openevse would then autoupdate. 

I've renamed the "Set" button to "Update", since the openevse time is never not set, it's just incorrect until updated.

jeremypoulter, 2017-08-16:
I don't think automatically updating the time whenever the browser is opened is necessarily a good idea. It was pointed out in #15 that the client browser may be in a different timezone for example.

Maybe we could just check and if the time is out by more than an hour open a pop up?

glynhudson, 2017-08-19:
Very true, although I would argue that it would be very unlikely that the client browser would be a different timezone. A popup notification would be a good solution. It's important that the time is set correctly or else the delay timer won't work as expected!

jeremypoulter, 2017-09-07:
I didn't think we had come to a conclusion on what to do with this issue. I did find another example (OpenWRT I think) that did the same thing. They had a button "Sync with Browser". We could change Automatic to Browser?

glynhudson, 2017-09-08:
Yes, good idea. Renaming the button to "Sync with Browser" makes a lot of sense.

commit d5d84756 (2017-10-10): change time update button Update > Sync fixes fixes #76
closed 2017-10-12 by glynhudson