OpenEVSE/ESP8266_WiFi_v2.x #77: Data corrupted after SPIFFS HTTP upload failure

glynhudson, 2017-08-16:
Just tried to upload SPIFFS via HTTP which seemed to fail now the unit does not seem to be responding:

Does this debug log give you a clue as to what might have gone wrong? 

```
e: 10736
Free: 13320
Time1
emonpi/emoncms/input/post.json?node=openevse-http&json={amp:0,wh:0,temp1:305,temp2:0,temp3:230,pilot:32,state:254,freeram:13176,divertmode:1}&apikey=906b00dc816304b4793568913d1c14c2
HTTP
ok
openevse/amp = 0
openevse/wh = 0
openevse/temp1 = 305
openevse/temp2 = 0
openevse/temp3 = 230
openevse/pilot = 32
openevse/state = 254
openevse/freeram = 13176
openevse/divertmode = 1
Free: 18808
Free: 18808
Free: 18808
Free: 18808
Free: 18808
Free: 18808
Free: 18808
Free: 13320
Free: 18808
Free: 18808
Free: 18808
Free: 18808
Free: 18808
Free: 18976
Free: 20896
Time1
emonpi/emoncms/input/post.json?node=openevse-http&json={amp:0,wh:0,temp1:305,temp2:0,temp3:230,pilot:32,state:254,freeram:20872,divertmode:1}&apikey=906b00dc816304b4793568913d1c14c2
HTTP
ok
openevse/amp = 0
openevse/wh = 0
openevse/temp1 = 305
openevse/temp2 = 0
openevse/temp3 = 230
openevse/pilot = 32
openevse/state = 254
openevse/freeram = 20872
openevse/divertmode = 1
Free: 20896
Free: 20896
Free: 20896
Free: 20896
Free: 20896
Free: 17360
Free: 16096
Free: 20344
Free: 14584
Free: 14584
Free: 19240
Free: 19240
Free: 19240
Free: 11120
Time1
emonpi/emoncms/input/post.json?node=openevse-http&json={amp:0,wh:0,temp1:305,temp2:0,temp3:230,pilot:32,state:254,freeram:11144,divertmode:1}&apikey=906b00dc816304b4793568913d1c14c2
HTTP
Emoncms error: server error: -1
openevse/amp = 0
openevse/wh = 0
openevse/temp1 = 305
openevse/temp2 = 0
openevse/temp3 = 230
openevse/pilot = 32
openevse/state = 254
openevse/freeram = 11144
openevse/divertmode = 1
Free: 11008
Free: 10984
Free: 10960
Free: 11104

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

Using default ''
Got '�' 15 == 110 @ 544:33
Using default ''
Got '' 255 == 128 @ 128:45
Using default 'data.openevse.com/emoncms'
Got '' 255 == 128 @ 173:32
Using default 'openevse'
Got '�' 0 == 4 @ 438:60
Using default '7D:82:15:BE:D7:BC:72:58:87:7D:8E:40:D4:80:BA:1A:9F:8B:8D:DA'
Got '' 255 == 128 @ 205:45
Using default 'emonpi'
Got '' 255 == 128 @ 250:32
Using default 'openevse'
Got '' 98 == 128 @ 282:32
Using default 'emonpimqtt'
Got '(a,b){var c,d,e=n.ajaxSettings.flatOptions||{};for(c in b)void ' 48 == 172 @ 314:64
Using default 'emonpimqtt2016'
Got '!==b[c]&&((e[c�' 105 == 127 @ 378:30
Using default ''
Got 'b.js' 0 == 213 @ 408:30
Using default ''
Got '�' 0 == 63 @ 498:16
Using default ''
Got '�' 0 == 80 @ 514:16
Using default ''
Got '�' 0 == 97 @ 530:10
Using default ''
Got 'eb00ec'  0 == 135 @ 540:4
Using default '000000'
After config_load_settings: 30760
Starting AP
AP IP Address: 192.168.4.1
After wifi_setup: 29504
Server started
After web_server_setup: 23456
After ota_setup: 22560
Free: 22640
first read RAPI values
Free: 22872
Free: 22872
Free: 22952
Free: 22952
```

jeremypoulter, 2017-08-17:
>Using default 'emonpimqtt'
> Got '(a,b){var c,d,e=n.ajaxSettings.flatOptions||{};for(c in b)void ' 48 == 172 @ 314:64

Hum the config looks to have been overwritten... I will check on this.

glynhudson, 2017-08-17:
Yes, I think this is what happened. When I reloaded the flash and SPIFFS manually the config had been lost

glynhudson, 2017-09-05:
After re-flashing using serial I've just tried again to update the spiffs via http and experienced the same issue:

� characters being printed out at startup being retrieved from EEPROM

jeremypoulter, 2017-09-18:
Still an issue but SPIFFS is no longer needed due to #87