OpenEVSE/ESP8266_WiFi_v2.x #80: Don't save the max current during divert

jeremypoulter, 2017-08-23:
The dev firmware allows you to specify if the max current is saved.

'''
SC amps [N]- set current capacity
response:
if amps < minimum current capacity, will set to minimum and return $NK amps
if amps > maximum current capacity, will set to maximum and return $NK amps
otherwise return $OK amps
default action is to save new current capacity to EEPROM.
if N is specified, then new current capacity is volatile, and will be
reset to previous value at next reboot
'''

When changing the max current during divert the value should not be saved.

Question how is this handled in the stable firmware.

fab33, 2017-08-23:
I check the code for mqtt and all the payload seems to be sent to the serial port. So I think it works.

jeremypoulter, 2017-08-25:
Yes, when sending the RAPI commands directly via MQTT or HTML then the RAPI is just passed through directly. So that is good that meets your requirements. However the divert code also sets the current so needs to use this feature so I will leave this issue open for that.

glynhudson, 2017-09-04:
Ah, that's fantastic. If your running the dev FW does this mean if you set the current using `$SC 16 N` does this man the current will be temporally set to 16A for the current charging session? Then revert back the the default max current once the car is unplugged? This would be very useful. 

We hope to ship all new OpenEVSE units with the new FW, dev is very soon to become the new stable

fab33, 2017-09-05:
@glynhudson yes you are right setting $SC 16 1 (1 if you don't want to store value). I use this to save flash from arduino (limited write). I have a node-red flow to limit current if I use other appliance in my house. My Electric meter report total usage and I dynamicaly adapt my charging current. Actually I don't use your wifi firmware (using ESP-link) but I investigate for a new firmware.

jeremypoulter, 2017-10-16:
Fixed by #103

PR #103 (2017-10-13): Changes to not save the current in EEPROM
Fixes #80
commit 9cceb100 (2017-10-13): Changes to not save the current in EEPROM, Fixes #80
closed 2017-10-16 by jeremypoulter