OpenEVSE/ESP8266_WiFi_v2.x #85: Print IP address to LCD on WiFi connect

glynhudson, 2017-09-06:
After a hard reset I connected to the ESP WiFi AP and entered my SSID password. The Wifi ESP connected successfully. It would be very useful if at this point the ESP printed to the openEVSE LCD to announce successful connection (or otherwise!) it's new IP address on the network like it does at startup.

jeremypoulter, 2017-09-06:
It should do this already, however I have already made some more
improvements to the flow/LCD display on my install_improvements branch.

glynhudson, 2017-09-06:
Awesome! Thanks. I'm pretty sure it doesn't currently, I was videoing the
lcd at the time with thoughts to make a setup video

- sent from my mobile device

jeremypoulter, 2017-09-18:
Fixed as part of #90

closed 2017-09-18 by jeremypoulter