OpenEVSE/ESP8266_WiFi_v2.x #93: Check safety error reporting

glynhudson, 2017-09-19:
Sean has reported that the safety errors don't get recorded in the UI

https://community.openenergymonitor.org/t/openevse-with-wifi-build-v2-and-testing/4993/13?u=glyn.hudson

jeremypoulter, 2017-09-19:
http://openevse.local/config :

http://openevse.local/config :
http://openevse.local/config :

```
{“firmware”:“3.11.3”,“protocol”:“1.0.3”,“espflash”:4194304,“version”:“2.3.0”,“diodet”:0,“gfcit”:0,“groundt”:0,“relayt”:0,“ventt”:0,“tempt”:0,“service”:2,“scale”:220,“offset”:0,“gfcicount”:fe,“nogndcount”:2,“stuckcount”:0,“ssid”:“EE-79q7nc”,“pass”:“DUMMY_PASSWORD”,“emoncms_enabled”:true,“emoncms_server”:“emoncms.org”,“emoncms_node”:“30”,“emoncms_apikey”:“DUMMY_PASSWORD”,“emoncms_fingerprint”:"",“mqtt_enabled”:true,“mqtt_server”:“emonpi”,“mqtt_topic”:"",“mqtt_user”:"",“mqtt_pass”:"",“mqtt_solar”:"",“mqtt_grid_ie”:"",“www_username”:"",“www_password”:"",“ohm_enabled”:false}
```

“gfcicount”:fe

That is the problem, I think I need some hex to decimal conversions on the error counters.

glynhudson, 2017-09-22:
#97 has fixed this for me 👍

PR #97 (2017-09-22): RAPI error checking
Improved the error checking around RAPI command responses, including correctly parsing number responses and storing them as long rather than strings. This makes sure the JSON is always correctly generated, fixes #93.
closed 2017-09-22 by glynhudson