OpenEVSE/ESP8266_WiFi_v2.x #94: Verify in Arduino

roy13, 2017-09-20:
Thanks Glyn i saw the post on openenergymonitor but i have used my allowed 3 posts in one topic and have been cut off.
I assume the server_improvements is the github file to download eliminating memory problems. However when i compile on Arduino i get this.
two or more data types in declaration of 'event_send'
referring to line 174; void event_send(String event)

glynhudson, 2017-09-20:
Hi @roy13 

I've now updated your user on our forum to remove the limitation. 

@jeremypoulter can confirm but I beleive after the latest updates the FW will no longer compile using Arduino IDE. You will need to use PlatformIO

jeremypoulter, 2017-09-20:
@roy13, @glynhudson I was trying to keep Arduino IDE compatibility but I have not checked it recently. I will have a look and see if there is a quick fix.

PR #95 (2017-09-22): Arduino IDE fix
Fix for compiling in Arduino IDE, fixes #94 . Was due to Arduino adding its magic in the wrong place. Also added building with the Arduino IDE to the Travis CI builds to help maintain Arduino IDE compatibility
closed 2017-09-22 by glynhudson