OpenEVSE/ESP8266_WiFi_v2.x #98: MQTT failing to connect causes lockup

glynhudson, 2017-09-22:
If MQTT fails to connect (MQTT server is down) this causes the openevse to lock-up and web-page hangs at "loading"

```
OpenEVSE WiFI 1383250
Firmware: 2.3.0
Free: 31160
Connecting to SSID: Crumble
Connected, IP: 192.168.1.102
Server started
Free: 27368
MQTT Connecting to...emonpi

ng to...emonpi
MQTT failed: -2
MQTT Connecting to...emonpi
MQTT failed: -2
MQTT Connecting to...emonpi
MQTT failed: -2
MQTT Connecting to...emonpi
MQTT failed: -2
MQTT Connecting to...emonpi
MQTT failed: -2
MQTT Connecting to...emonpi
MQTT failed: -2
MQTT Connecting to...emonpi
```

jeremypoulter, 2017-09-22:
This will largely be down to the MQTT library being blocking, should be fixed by implementing #30

glynhudson, 2018-01-29:
Does not seem to be an issue anymore

closed 2018-01-29 by glynhudson