OpenEVSE/ESP8266_WiFi_v2.x #100: web access hangs when accessed from multiple devices?

ebeighe, 2017-09-23:
Here is what i think i've seen:
With the web UI open and everything working normally... at some random point the web UI seems to hang (though that isn't apparent in the UI) and from then on *all* web requests seem to hang (even just a single command, e.g. http://openevse/r ).

My trouble seems to stem from me leaving another computer also open to the web UI. If i take care to only every have one computer open to the web UI everything seems to work fine indefinitely.
To clear the problem, I think I have to hard reset the unit with a power cycle. 

Even  while the problem is otherwise occurring, emoncms data is still flowing as normal, and the unit can be pinged just fine.

jeremypoulter, 2017-09-23:
Which version of the firmware are you using, the master branch has significant fixes that help with this type of issue.

ebeighe, 2017-09-24:
According to the system tab, i have:
WiFi Firmware ESP8266 Version: 2.3.0
I have whatever firmware came with the wifi module from openevse, i purchased just a couple weeks ago.

I must confess I don't really know much about the code, and am not planning on trying to compile it.
Is there a place to get the firmware that is ready-to-load through the web gui?

I am a little confused about the versioning scheme; is it correct that the version that appears in the web gui is BUILD_TAG in the file platformio.ini?
I would think i have "stable" firmware (i.e. from the factory), but "stable" is BUILD_TAG=2.1.0; whereas in "master" it was bumped to 2.3.0 since Aug 17th.
How does one know more exactly what code one has?

Sorry! i know these are very basic questions... Thanks!

ebeighe, 2017-09-24:
oops, didn't really mean to "close".

glynhudson, 2017-09-25:
Hi @ebeighe, 

The WiFi ESP FW is about to get a big release. Yes, the version shown in the UI is the platform IO build tag. However, since we have been working on a major update in private (without releasing) we have been lapse in updating the version...until now when we're preparing for a big update release. 

I have just released version V2.4.0, please could you try and update (upload firmware.bin via web interface) and let us know it fixes this issue: 

https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/releases/tag/2.4.0

ebeighe, 2017-09-25:
Ah, great.
I flashed the firmware.bin from the web gui and that worked fine.
So far so good, i have it open in three browser windows across two computers for and hour or so and everything's still working, not getting "stuck" :-)

Thanks!

closed 2017-09-25 by ebeighe