OpenEVSE/ESP8266_WiFi_v2.x #101: HTTP security causing web-page loading hang

glynhudson, 2017-09-27:
Recently I enabled HTTP authentication on the openevse. Occasionally the webpage will hang when trying to login. Just wondering if this could be replicated?

@jeremypoulter have you tested with HTTP authentication enabled?

tgould, 2017-10-05:
I have experienced this issue and can replicate.  Happens when you use 2 different devices to access the web UI (may only occur if within an Auth Timeout period.)  The first will work and then every attempt after from any other device will fail.  On 2.4.1.  With HTTP Auth fields cleared out, I do not experience it.

glynhudson, 2017-11-14:
I vote removing HTTP auth to fix this issue since basic HTTP auth over non-SSL connection is basically worthless anyway and could give a user a false sense of security.

chris1howell, 2017-11-14:
I do not agree, security is achieved by adding layers. HTTP auth is only
useless if someone has enough access to capture and decode packets. Over
the air we have WPA. Over the wire the network should be physically secured
especially in a home. A milicious party would have to compromise a device
that is capible enough to capture and decode packets then catch an
authentication in progress.

Disabling auth would allow RAPI commands to be sent and exicuted without
all the work to compromise the network, capture the password during
authentication then send authenticated packets.

Hopefully a HTTPS server becomes available and then we do not have to worry
about unencrypted authentication.

glynhudson, 2017-11-17:
Valid points @chris1howell. Let's try and fix it rather than remove it.

PR #125 (2017-11-24): 'Fix' for auth crashing the ESP
I use the term fix very loosely. What this actually does is allow the StaticFileWebHandler to have the authentication separately configured, as with the other handlers, and then not configure authentication.

The effect of this is that only the dynamic content asks for authentication. This should be ok but the underlying bug in the new StaticFileWebHandler is still there.

Fixes #101
commit a281efaf (2017-11-24): Merge pull request #125 from jeremypoulter/auth_fix fixes #101

'Fix' for auth crashing the ESP
closed 2017-11-24 by glynhudson