OpenEVSE/ESP8266_WiFi_v2.x #107: Different timezones causing elapsed time to be incorrectly displayed

glynhudson, 2017-10-13:
As reported:

https://community.openenergymonitor.org/t/openevse-wifi-gateway-fw-stable-release-v2-4-1/5340/2?u=glyn.hudson

I notice in your screenshots that your Elapsed Time always has leading zeros. My Elapsed Time always leads with 19. I live in Atlanta, GA. I am assuming this is due to GMT-5 Eastern Time Zone? Screenshot below: (I was charging for 4 min 21 sec. not 19 hours 4 min and 21 sec.)

[image]

joverbee, 2018-01-06:
I think I have solved this one..the root cause seems that the elapsed time is calculated as a difference of two date variables and then applying dt.getHours() on it. However, this applies an extra correction for the time zone which is unwanted here (a time difference is a time difference irrespective of the time zone). Replacing this with dt.getUTCHours() overrides this correction and leads to the correct session time (at least for me in Belgium). (affects src/html/view_models/TimeViewModel.js)
will try to do a pull request, can also send a binary file for quick checking by others if useful.

jeremypoulter, 2018-01-06:
Good find, let me know if you need help with creating the PR

joverbee, 2018-01-06:
i tried, but seem to get stuck disentangling 2 commits for unrelated issues. (I really need to upgrade my github skills, its not the first time I get confused about this)

glynhudson, 2018-01-10:
Nice spot. A simple way of doing a PR is to edit the file directly on
github.com or clone a fresh version then make your changes. Alternatively
just email me a copy if the file with your changes, I'll test, do a diff
and open a pr.

Thanks a lot.

- sent from my mobile device

glynhudson, 2018-01-28:
Confirmed fixed by #132. Thanks @joverbee 👍