OpenEVSE/ESP8266_WiFi_v2.x #109: IP address not printed correctly to LCD at startup

glynhudson, 2017-10-13:
This was a known issue with older FW, however I've just tested with latest openevse V4.8.0 controller FW and the IP address is not printed correctly to LCD at startup. Presumably because openevse controller is printed to the LCD. See video:

https://www.youtube.com/watch?v=R5cj0Y-uQSI

jeremypoulter, 2017-10-13:
I have seen this I think it only happens when both units are starting up at the same time.

glynhudson, 2017-10-13:
I guess unfortunately  both units will always startup at the same time when
the power to the unit is switch on. Maybe we could add a delay in the wifi
FW to delay posting the IP address until openevse has finished stating up

- sent from my mobile device

closed 2017-11-03 by glynhudson