OpenEVSE/ESP8266_WiFi_v2.x #111: HTTPS/SSL Fix Ohmconnect

glynhudson, 2017-10-19:
@chris1howell notes that the Ohmconnect integration with openevse wifi is currently broken since the hardcoded fingerprint has expired: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/blob/stable/src/ohm.cpp#L16

[ohmconnect](https://www.ohmconnect.com/) API is HTTPS only. This is far from ideal to have to hard code a fingerprint that expires every 3 months or so. 

There is an open PR to add SSL support to Arduino esp8266 core: https://github.com/esp8266/Arduino/pull/3001 . Once this is merged (or before) we could test proper HTTPS connections. If this works well it would be great to make HTTPS default for Emoncms

jeremypoulter, 2017-11-02:
I think this needs HTTPS client support as detailed in #44 rather than server support, good news is this has already been merged.

jeremypoulter, 2017-11-02:
One thing about also applying to EmonCMS, is that this will be limited to connecting to HTTPS servers that use Let's encrypt certs (or rather DST). It will not be as easy to change the cert as it is the current finger print.

glynhudson, 2017-11-03:
The cert for Emoncms.org is from letsencrypt and the fingerprint changes every 30 days when the cert is renewed. What is DST?

With regard to Emoncms we @TrystanLea was implemented a secure API over HTTP which uses the API to encrypt the data. 

`URL: /input/post or /input/bulk HEADER: Authorization: USERID:HMAC_HASH, Content-Type: aes128cbc
POST BODY: IV+CIPHERTEXT`

See encryption section of: http://emoncms.org/site/api#input

chris1howell, 2017-11-03:
What do you guys think about setting up our own sudo authority?

-put the fingerprints for data.openevse.com<http://data.openevse.com>, emoncms.org<http://emoncms.org> and ohmconnect.org<http://ohmconnect.org> on a server we control
-I have a static IP and I could setup a virtual server for this purpose.
-file would be in JSON format
-ESP would connect to server by IP Address (to avoid DNS redirect flaws)
-Download the file periodically
-When fingerprints change we modifiy the file on the server and the firmware does not need to change.

jeremypoulter, 2017-11-04:
>The cert for Emoncms.org is from letsencrypt and the fingerprint changes every 30 days when the cert is renewed. What is DST?

DST provide the root certificate for Letsencrypt. 

I will take a look at the new protocol, shouldn't be two difficult to integrate, although I have to say I would much prefer to use HTTPS over something new and relatively untested (from a security POV)

chris1howell, 2018-03-24:
Until we get proper SSL, this is an alternative to hardcoding  or saving the fingerprints to EEPROM.

Below is a quick python script that will look at each certificate and output the fingerprint to a JSON file. I can host this on a server. From the ESP8266 we could download the file directly by IP on boot, then any time there is a fingerprint mismatch.

Here is the output in JSON.

 >{ "data.openevse.com":"8a0a5eef29033e2fc90fd12023f015c76e23f7ce","emoncms.org":"8a0a5eef29033e2fc90fd12023f015c76e23f7ce","ohmconnect.com":"8a0a5eef29033e2fc90fd12023f015c76e23f7ce","owner-api.teslamotors.com":"8a0a5eef29033e2fc90fd12023f015c76e23f7ce" }

import ssl
import socket
import hashlib
 
addr = 'data.openevse.com', 'emoncms.org', 'ohmconnect.com', 'owner-api.teslamotors.com',
json = "{ "
 
sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
sock.settimeout(1)
wrappedSocket = ssl.wrap_socket(sock)
 
for i in range(len(addr)):
  try:
    wrappedSocket.connect((addr[i], 443))
  except:
    response = False
  else:
    der_cert_bin = wrappedSocket.getpeercert(True)
    pem_cert = ssl.DER_cert_to_PEM_cert(wrappedSocket.getpeercert(True))
    thumb_sha1 = hashlib.sha1(der_cert_bin).hexdigest()

  wrappedSocket.close()

  json = (json + '"' + addr[i] +'":"' + thumb_sha1 + '",')

json = json[:-1]
json = (json + " }")
print json

jeremypoulter, 2018-03-24:
Good idea in theory, but you have to protect (ie digitally sign) the JSON file, so you have unfortunately just moved the problem.

chris1howell, 2018-03-24:
None of the data is particularly sensitive. Which is why I suggest pulling the file by static IP instead on DNS. This would give a reasonable amount of assurance that the file is from the correct source and avoid DNS redirection vulnerabilities.

It would be fairly complex to both spoof the JSON file and redirect to ESP to invalid hosts with spoofed fingerprints.

chris1howell, 2018-03-24:
I see it as a step in the right direction. Right now almost 100% of users are using http and ohmconnect is broken beacuse it is https only.

So what is better http with no protection or downloading the fingerprints and encrypting everything...? Sure it's not perfect, full SSL is best but this gets us 80% there.

Would have to disagree. The data is sensitive in that if you intercepted
and replaced the fingerprints you can then intercept the HTTPS traffic
which nullifies the point of having the HTTPS.

Jeremy

kicker22004, 2018-12-30:
Was this ever fixed? I'm trying to get mine connected but can't get it to. I don't mind hardcoding it if I have to. But where do I find the fingerprint up replace the set one.?

kicker22004, 2018-12-30:
Ok I'll give it a test later today.

glynhudson, 2019-01-02:
Did the code compile correctly? What error did you see? 

Here is the latest master branch compiled as a .bin (in a zip file) 

[firmware.zip](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/files/2722379/firmware.zip)

kicker22004, 2019-01-02:
I'm not sure I did it correctly, I compiled it with Arduino Ido for the esp8266. It compiled correctly made the bin but would fail.

glynhudson, 2019-01-03:
Ah, I think you need to use PlatformIO to correctly compile

kicker22004, 2019-01-03:
I had platform IO installed and it compile correctly. I flashed my esp8266 with it and it worked but when I compile a bin it will fail to upload.

kicker22004, 2019-01-03:
I was attempting to update with the .bin via the WebUI section. Is this wrong?

kicker22004, 2019-01-03:
> Interesting, could you post the error you see? Can you upload my .bin using esptool?

Ok so your zip worked... Is it because mine wasn't in a zip file?

Also under my WiFi Firmware it says Version: BUILD_TAG instead of build version. This is on my personal board the Wemos D1 R1 I'm using it as testing before updating my official Board in the charger. So i'm compiling via Android 1.8.2 > Sketch> Export Compiled Binary.

kicker22004, 2019-01-03:
Even with your build i'm unable to get Ohmconnect to see my device.

https://us.v-cdn.net/6028386/uploads/editor/y6/f6pk9tovnt7u.jpg

chris1howell, 2019-01-03:
It looks like you are connected to ohmconnect there is just no OhmHour right now which is why it says "None".

https://us.v-cdn.net/6028386/uploads/editor/y6/f6pk9tovnt7u.jpg

chris1howell, 2019-01-03:
> Ok I just don't see my device being listed on ohmconnect's website is why I believed it didn't work.

Open Source devices are "listen only" on OhmConnect. Once a minute OpenEVSE checks for an OhmHour. It will report back "None" when there is no Ohm Hour or "True" when there is. OpenEVSE will go to sleep during OhmHours.

kicker22004, 2019-01-03:
Well that explains that then. So looks like it's working then. Thank you for all your help.

glynhudson, 2019-01-21:
The fix has been included in stable release V2.7.7

https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/releases

closed 2019-01-21 by glynhudson