OpenEVSE/ESP8266_WiFi_v2.x #112: Display cost per session on web interface & LCD

glynhudson, 2017-10-19:
With the new RAPI API to pause OpenEVSE LCD controller updates it would be possible to alternate the LCD display between the standard LCD display (session energy & total energy) from the openevse controller and cost of charge (session & total) in local currency from WIFi ESP over RAPI API.

- Set unit cost via web interface 
- Set currency symbol via web interface
- Ability to update unit cost via MQTT to enable smart dynamic pricing
- Display session cost & total cost on web-interface 
- Alternate LCD between session energyt & total energy (default) and session cost & total cost (printed from ESP WiFi)

Do you agree @chris1howell? Any further ideas?

cmgj74, 2017-12-19:
I would like this. :)

jeremypoulter, 2018-03-06:
This begs the question of how this would interact with PV divert, I presume those with PV divert enabled should have that deducted from the total and they would like to know the amount saved?

Do you have an example of a feed that you would provide over MQTT? Is that the most appropriate source, eg is there not a online resource for this?

glynhudson, 2018-03-11:
Good point. 

When solar PV divert is active we know the current generation of the solar PV, this generation power should could be subtracted from the current charging power when calculating cost.

However, to avoid getting blogged down in detail (for first implementation of cost feature) I would not be adverse to just disabling cost when solar PV is active as we can probably assume that most power is coming from solar PV. I think we can assume that users who will be using solar PV divert will probably have Emoncms setup to record cost of grid import / export since an emonPi / MQTT setup is required to provide the solar PV gen feed in the first place. 

I see charging cost feature being more of use when the unit is installed in multi user / public charging location where solar PV divert would probably not be used. e.g. multi occupancy flats sharing the same charger, cafe/business offering charging to the pubic. The cost feature will allow each user to see on the display how much each charge has cost easily on the LCD. 

I'm keen for cost rate to be possible to adjust dynamically via MQTT to support real-time pricing. The current charge rate should be displayed on the LCD. The LCD should alternate between the normal energy display and cost display (if this is possible). 

Any other thoughts on this feature @cmgj74 @jeremypoulter @davehun @TrystanLea