OpenEVSE/ESP8266_WiFi_v2.x #114: Implement OCPP protocol V1.6

glynhudson, 2017-10-23:
Open Charge Point Protocol: http://www.openchargealliance.org/protocols/ocpp/ocpp-16/

Allow OpenEVSE to integrate easily with other charging networks.

chris1howell, 2018-01-08:
Here is a white paper about integrating OCPP with OpenADR (Demand Response) both would fit in well with EmonCMS energy monitoring to create Smart Charging with open protocols.
[using openadr with ocpp.pdf](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/files/1613079/using.openadr.with.ocpp.pdf)

DrFrankReade, 2018-04-04:
This is a great idea, but is the ESP8266 powerful enough to get the job done? Or is this a good job for the Raspberry Pi Zero W or an ESP32?

glynhudson, 2018-04-04:
Yes, I think your right. OCPP is a hefty API. More horsepower will be required. Also for https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/152

My preference would be RaspberryPi, possibly a zero or A+

DrFrankReade, 2018-04-05:
I have friends who swear by the ESP32, and it's an awesome chip for sure,  but code for a Pi, being (likely) linux, is going to be a LOT more portable than the code for an ESP32. So, there's nothing I love more than designing PCBs - Shall I take a stab at a board to mate the OpenEVSE to a Pi Zero? With level shifters and transient protection?  My only concern is powering the Pi. I'm not sure if the OpenEVSE's power supply has the oomph to power it, certainly not from the 5V rail, but maybe the 12V rail through a buck converter.  I'll look in to it.

glynhudson, 2018-04-05:
It defiantly need it's own power supply. 

That's a very kind offer. However, I'm not sure we need a PCB since the openevse controller can just talk to the Pi via serial. A wire to the GPIO serial pin is all that's needed. 

@chris1howell @lincomatic @jeremypoulter do you have any thoughts?

chris1howell, 2018-04-05:
I really like the OrangePi Zero. It is nice and small and has both WiFi and Ethernet.

OpenEVSE serial is 5v as long as the pins on whatever SBC we connect to are 5v tolerant we are okay. Otherwise we need a level shifter.

We will absolutely need external power unless we beef up the PS on a future version.

jeremypoulter, 2018-04-05:
I think an ESP32 port should be fairly easy. Just need a few hours to have
a look as far as I can tell all the required libraries are supposed.

For the Pi/Orange Pi/basically any PC with a serial port I kind of had in
mind that the Simulator could be relatively easily modified to work with
the OpenEVSE directly. I have a few ideas on how to do this.

I am nearly done with the time zone issues, it may be an idea to look at
porting to something with a bit more power.

https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/114#issuecomment-378795496>,

DrFrankReade, 2018-04-05:
Hello -

One could also simply plug a USB FTDI adapter in the Pi and go that way, but the pile of hardware is starting to look really ugly, which is why I advocate for a purpose built board. 

On the matter of serial between the OpenEVSE and a Pi, or ESP of any variety, level shifting is absolutely needed. Although a simple voltage divider will suffice for getting data from OpenEVSE to a 3.3V device, a proper level shifting is required to get the data up to 5V for the OpenEVSE to read any of these devices.  There's a  dozen ways to do this, but I'm personally a big fan of using optoisolators because of the dual purpose that they'd be serving in this role.  While it *might* work to directly connect it, reliability often ends up suffering. 

Another reason I like a board is to be able to power the Pi from a more robust connector than the micro USB, and the possibility of powering it with a higher voltage, maybe even the 220 lines (The MeanWell IRM-10-5 comes to mind) without having to incorporate a wall wart somewhere.  I see the whole shootin' match to look like a raspberry pi zero W stacked and screwed to a board with 220V in and 5V serial out (and maybe a couple LEDS)

chris1howell, 2018-04-05:
I am pretty sure the Atmega 328p will handle 3.3v tx levels even at 5v.

DrFrankReade, 2018-04-05:
Hello,

It's true, the ATMEGA328P can in fact handle 3.3V as a logic "high" but since the "high" on the atmel is defined as VCC * 0.6 (that's 3V), there's not much room for power supply droop or noise, The adafruit boards work because they have shifters built in. 

Andy

davehun, 2018-04-06:
A board with more CPU grunt and connectivity would help for other reasons.
A Board with BT and wifi would allow getting the state of charge via ODBC i.e. a leafspy type device.
I haven't had a good experince getting the state of charge via the Nissan API (other cars may be better).

glynhudson, 2018-04-12:
Having been involved in installing a few OpenEVSE units with WiFi in customer locations I'm convinced of the importance of having a wired Ethernet option. In some locations WiFi is tricky and can be unreliable when the unit is mounted externally. 

I'm keen to stick with an embedded solution rather than move to a full linux stack e.g RaspberryPi. I feel that an embedded solution e.g. ESP32 will be more reliable and better suited to the working environment inside an openEVSE enclosure e.g wide temperature range etc. 

I've found that it's possible to get a board with an ESP32 with integrated Etherned which to me seems to perfect solution: https://www.olimex.com/Products/IoT/ESP32-GATEWAY/open-source-hardware

ESP32 will enable secure HTTPS connections and give us plenty of space for growth.

What do you think @jeremypoulter @chris1howell @lincomatic?

jeremypoulter, 2018-04-12:
Oh, that looks like a nice board. I have started on a port to ESP32,
unfortunately (or fortunately depending on your POV) I need to implement
propper SSL certificates.

Honestly thought I don't think there is any advantage, from a hardware POV,
in using the ESP32 over a Pi (or similar). Certainly for large scale,
remote management applications I would be more inclined to go for the
latter.

In essence I think there are good use cases for each of the platforms.

DrFrankReade, 2018-04-12:
I agree that wired ethernet is always better if it can be pulled off. Wifi is problematic in dense areas, even if the signal looks "good"

I will again make the case for a Pi (or other Linux device) since there are TONS of them, and portability of code will (practically) never be an issue. Run it on a Pi, an Intel Edison, a PC, an Arduino Yun, etc.. and there are linux distributions that are lightweight and boot quickly. On the other hand, if for some reason they stop making ESP32s, then that's just that.  There are also infinite peripheral options on the Pi and fantastic community support (the ESP has a good community too)
Imagine it... [http://www.frank-buss.de/raspberrypi/epaper/energy-monitor-display.jpg]
(for a PowerWheels EVSE)

chris1howell, 2018-04-17:
OCPP 2.0 is out.
[OCPP-2.0.zip](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/files/1921700/OCPP-2.0.zip)

DrFrankReade, 2018-05-24:
Somebody's been working on this...
http://ibeyonde.com/wordpress/ocpp-compliant-openevse/

glynhudson, 2018-05-25:
Interesting, I will try and make contact. Thanks for the link. Sadly it doesn't look like they will be very open-source with their implementation.

DrFrankReade, 2018-05-25:
The big question here is if a Docker or Raspberry Pi image exists with a nice beautiful OCPP server with a web interface for people to use (no of course not) Or if it's python from the command line. I wish I was smarter and could make these things.  Some of these interfaces that people are making are so damn nice (The PiHole comes to mind, even EMONPI is pretty slick)

jeremypoulter, 2018-05-25:
There are a few Node JS implementations that could be useful in conjunction with #183.

- https://www.npmjs.com/package/ocpp-js

Precisionsflyg, 2018-05-30:
I would also like to see OCPP support, but more importantly a lightweight OCPP server on RPi.
I'm weighing my options right now, and the EmonEVSE looks like a good fit, but since I would like to also put a few up at work, it's missing an easy way to keep track of energy usage. It's also missing an RFID reader, to be able to track who uses which charger - but thats another problem... ;-)

jeremypoulter, 2018-05-30:
Thanks for showing your interest. 

The [Node.JS library](https://www.npmjs.com/package/ocpp-js) looked like it had a fairly light weight server, may meet your needs.

RFID is an interesting idea, obviously there is a bunch of business logic here and we would need the right split to allow folks to integrate to whatever access control/charging (money) system the have in the backend. You should raise a ticket for that so we don't go too off topic here.

chris1howell, 2018-05-30:
I agree, we need 3 components...

OCPP 2.0 client
OCPP 2.0 Server to log to database
User Interface for Energy Monitoring and Control (maybe OCPP UI plugin for EmonCMS?)

chris1howell, 2018-05-30:
Unfortunately, the Node.JS library<https://www.npmjs.com/package/ocpp-js> is a bit outdated. It is based on version 1.5 SOAP. Version 1.6 has been out for quite a while and uses SOAP or JSON. Version 2.0 was recently released.

On the server side I think SteVe is a better option. It already supports OCPP 1.6 JSON.
https://github.com/RWTH-i5-IDSG/steve

On Wed, May 30, 2018 at 8:55 AM, chris1howell . <chris1howell@msn.com<mailto:chris1howell@msn.com>> wrote:
I agree, we need 3 components...

OCPP 2.0 client
OCPP 2.0 Server to log to database
User Interface for Energy Monitoring and Control (maybe OCPP UI plugin for EmonCMS?)

chris1howell, 2018-05-30:
I have a OCPP server up and running that can do OCPP 1.2, 1.5 and 1.6 SOAP and JSON. it is based on SteVE. https://github.com/RWTH-i5-IDSG/steve

You guys are welcome to check it out and test it with the OCPP client.

http://energy.openevse.com:8080/steve/manager/home

to hit the server via 1.6 JSON

ws://energy.openevse.com:8080/<http://energy.openevse.com:8080/>/steve/websocket/CentralSystemService/<chargeBoxId>

Have fun...

[cid:ii_jhtlg9xt0_163b2d92645a6827]

[https://plus.google.com/u/0/_/focus/photos/public/AIbEiAIAAABECMGOguW5qIypjQEiC3ZjYXJkX3Bob3RvKigxZTg5ODZmN2Q5MmViYTJiNDllMzU0NGVlZjliOGUwNWU2MzAzZDE0MAEAR3UQiVgVxBDxezvT7W2qShM1Jw?sz=24]
Click here to Reply or Forward

90.55 GB (88%) of 102 GB used
Manage<https://www.google.com/settings/u/0/storage?hl=en>
Terms<https://www.google.com/intl/en/policies/terms/> - Privacy<https://www.google.com/intl/en/policies/privacy/>
Last account activity: 3 minutes ago
Details

Page 1 of 21

On Wed, May 30, 2018 at 9:08 AM, chris1howell . <chris1howell@msn.com<mailto:chris1howell@msn.com>> wrote:
Unfortunately, the Node.JS library<https://www.npmjs.com/package/ocpp-js> is a bit outdated. It is based on version 1.5 SOAP. Version 1.6 has been out for quite a while and uses SOAP or JSON. Version 2.0 was recently released.

On the server side I think SteVe is a better option. It already supports OCPP 1.6 JSON.
https://github.com/RWTH-i5-IDSG/steve

On Wed, May 30, 2018 at 8:55 AM, chris1howell . <chris1howell@msn.com<mailto:chris1howell@msn.com>> wrote:
I agree, we need 3 components...

OCPP 2.0 client
OCPP 2.0 Server to log to database
User Interface for Energy Monitoring and Control (maybe OCPP UI plugin for EmonCMS?)

chris1howell, 2018-05-30:
I have a OCPP server up and running that can do OCPP 1.2, 1.5 and 1.6 SOAP and JSON. it is based on SteVE. https://github.com/RWTH-i5-IDSG/steve

You guys are welcome to check it out and test it with the OCPP client.

http://energy.openevse.com:8080/steve/manager/home

to hit the server via 1.6 JSON

ws://energy.openevse.com:8080//steve/websocket/CentralSystemService/<chargeBoxId>

Have fun...
[image]

chris1howell, 2018-06-10:
Ras Pi W Zero would work for MQTT to OCPP. 

A board with WiFi, Ethernet and USB like the Ras Pi 3 or Orange Pi Zero could also act as a WiFi Access Point and connect via WiFi, Ethernet or cellular 3G/4G.

chris1howell, 2018-06-10:
Here is an OCPP implementation that looks pretty promising. There is an example for a client.

https://github.com/NewMotion/ocpp

DrFrankReade, 2018-06-11:
Hello,

My personal feeling is that all of the bridges to bridges will make for far too many points of failure, plus a lot of stuff to set up. MQTT is awesome, but it's another layer.  A pi or other SBC attached directly to the OpenEVSE (well... via a level shifter) seems like the fewest moving parts.  OCPP is also the future, and moving forward, OCPP is going to be on virtually every commercial machine, and every network that survives the next 5 years is going to support it.  

There can be plenty of debate about the benefits or pitfalls of moving the I2C display over to the Pi (or any display) and the freeing up of resources on the atmega. It may even make the little power supply happier, although the change in the brownout voltage fuse a few months ago seems to have really ended squirrely powerups.  With the added mountain of high speed GPIO, any number of displays could be added, and by moving it over to Linux, the project is far more accessible to the community. Not that there isn't a HUGE ESP community, because there is.

The interface could benefit too. The ESP interface works fine, and doesn't of course needs any bells and whistles, but having easier access to schedules, graphs, etc, would be pretty great, and you don't have to hook up emoncms if you don't want to.  As far as OCPP, I guess it could be implemented with the client and the server running on the Pi together for stand-alone setups. 

Also, imagine the delight of just editing the HTML files and trying them out immediately!

jippej, 2018-06-11:
Hi,

just some thoughts from my side, please ignore if I am totally wrong here.

very interesting topic in respect to OCPP and where to implement what. I noticed that main idea here is that existing hardware needs to be expanded for implementing the OCPP client. I have been professionally involved in managing to setup an OCPP server to connect to many different chargers. conclusion was that not the complete client protocol is implemented  and only needed/supported features are implemented in most OCPP capable chargers. 
As I am not to software savy, I was wondering if it was not possible to implement a leghtweight/custom 
 made OCPP2.0 on the ESP32 with the Arduino JSON library for example (OCPP 2.0 only supports JSON)? As said during our testing we found that many instructions we sent to the chargers just replied back "not supported".

chris1howell, 2018-06-11:
Success from LINUX OCPP Example Client
https://github.com/NewMotion/ocpp

Connected OCPP 1.6 registered and validated RFID tag.

[image]

chris1howell, 2018-06-12:
@jippej I do not think you are totally wrong... Basic functions such as Boot Notification, Heartbeat and status are not very complex. I think it is possible to get some functionality on the ESP8266 as the ESP8266 does support websockets. This is the JSON for Boot Notification.

`{"chargePointVendor":"OpenEVSE","chargePointModel":"Advanced v5","chargePointSerialNumber":"0002089","firmwareVersion":"4.8.0","meterType":"Internal"}`

Here is a full session with Boot Notification, Status and even a RFID tag authorization.

`[INFO ] 2018-06-12 07:31:32,012 de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint - New connection established: JettyWebSocketSession[id=67f8a1b7, uri=ws://127.0.0.1:8080/steve/websocket/CentralSystemService/0002098]

[INFO ] 2018-06-12 07:31:32,350 de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint - [chargeBoxId=0002098, sessionId=67f8a1b7] Received message: [2,"0","BootNotification",{"chargePointVendor":"OpenEVSE","chargePointModel":"Advanced v5","chargePointSerialNumber":"0002089","firmwareVersion":"4.8.0","meterType":"Internal"}]

[INFO ] 2018-06-12 07:31:32,351 de.rwth.idsg.steve.service.CentralSystemService16_Service - The chargebox '0002098' is registered and its boot acknowledged.

[INFO ] 2018-06-12 07:31:32,356 de.rwth.idsg.steve.ocpp.ws.pipeline.Sender - [chargeBoxId=0002098, sessionId=67f8a1b7] Sending message: [3,"0",{"status":"Accepted","currentTime":"2018-06-12T14:31:32.351Z","interval":14400}]

[INFO ] 2018-06-12 07:31:32,443 de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint - [chargeBoxId=0002098, sessionId=67f8a1b7] Received message: [2,"1","Heartbeat",{}]

[INFO ] 2018-06-12 07:31:32,446 de.rwth.idsg.steve.ocpp.ws.pipeline.Sender - [chargeBoxId=0002098, sessionId=67f8a1b7] Sending message: [3,"1",{"currentTime":"2018-06-12T14:31:32.444Z"}]

[INFO ] 2018-06-12 07:31:32,470 de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint - [chargeBoxId=0002098, sessionId=67f8a1b7] Received message: [2,"2","StatusNotification",{"connectorId":0,"status":"Available","errorCode":"NoError"}]

[INFO ] 2018-06-12 07:31:32,476 de.rwth.idsg.steve.ocpp.ws.pipeline.Sender - [chargeBoxId=0002098, sessionId=67f8a1b7] Sending message: [3,"2",{}]

[INFO ] 2018-06-12 07:31:32,483 de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint - [chargeBoxId=0002098, sessionId=67f8a1b7] Received message: [2,"3","Authorize",{"idTag":"12345678"}]

[INFO ] 2018-06-12 07:31:32,485 de.rwth.idsg.steve.ocpp.ws.pipeline.Sender - [chargeBoxId=0002098, sessionId=67f8a1b7] Sending message: [3,"3",{"idTagInfo":{"status":"Accepted","expiryDate":"2020-05-01T07:30:00.000Z","parentIdTag":"10025"}}]

[WARN ] 2018-06-12 07:32:02,359 de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint - [id=67f8a1b7] Connection was closed, status: CloseStatus[code=1000, reason=null]`

chuck-h, 2018-12-04:
Worth checking out BCIT project here:
http://files.mgrid-bcit.ca/download/evid.php

They have written open source C language clients bridging Clipper Creek and Schneider to OCPP.

DrFrankReade, 2018-12-04:
> Regarding @glynhudson <https://github.com/glynhudson> @DrFrankReade
> <https://github.com/DrFrankReade> comments on Wifi connectivity issues,
> has anyone considered HomePlug (powerline ethernet)? I understand this is
> the network layer used for ISO 15118 Plug and Charge between EVSE and
> vehicle, so it must be pretty tolerant to the noise levels in this
> environment.
> Advantage over wired ethernet is simplified wiring installation.

Hello,

So as I understand it, HomePlug greenPHY actually runs over the pilot
signal for DC charging, but I'm not sure of the exact details, or if it's
available during AC charging, because that line is supposed to have a PWM
signal for "available current" running over it during regular use.  Maybe
they superimpose data packets during the high or low sections of the wave?
Or maybe it's just off limits. But it's a thing for sure.

glynhudson, 2018-12-04:
Hi @chuck-h, thanks for your comments. Powerline Ethernet is a good suggestion. It would certainly add hardware cost but maybe this cost would be lower than the individual Ethernet installation cost. Interesting idea.

@DrFrankReade I'm pretty sure during AC charging the pilot signal is only used to carry the PWM J1772 pilot signal. It's a real shame that SOC data is not available to the EVSE during AC charging. This would be very useful for EVSE smart scheduling.

lincomatic, 2018-12-04:
1) Yes, during AC charging, the pilot is reserved for the PWM advertised current only. You're not allowed to superimpose any signal on top of it.

2) Why do we need to make a special adapter for Powerline Ethernet? Can't users just buy and external Powerline Ethernet adapter (assuming that the EVSE has wired Ethernet)

chuck-h, 2018-12-05:
Re: "during AC charging, the pilot is reserved for the PWM advertised current only"
Are you certain of this? I have not read the actual ISO 15118 standards but I am pretty sure they anticipate communications continuing during charge. Since HomePlug frequency spectrum is 2 – 28 MHz there is no technical challenge in avoiding interference with the PWM pilot.

chuck-h, 2018-12-05:
@lincomatic is correct that however ISO 15118 uses HomePlug on the pilot line, it is irrelevant to this thread.

The HomePlug usage which is on-topic is to establish network communications with the EVSE. This can certainly be ordinary HPAV modules (no need for GP GreenPhy) like https://www.amazon.com/TP-Link-Powerline-ethernet-Adapter-TL-PA4010KIT/dp/B00AWRUICG.

The main discussion here seems to center on running OCPP1.6J bridge software on an ESP8266 or other embedded processor co-located with the EVSE.

Another plausible approach to consider would be to run native OpenEVSE serial over UDP or TCP to a separate processor which runs the bridge software. At the EVSE end, an off-the-shelf module like https://www.digikey.com/product-detail/en/netburner-inc/SBL2E-100IR/528-1016-ND/2521663 would work (again, watch the 3.3V signal levels), with a short ethernet jumper to an HPAV adapter. Is there a cheaper off-the-shelf solution?

chuck-h, 2018-12-05:
@chris1howell FYI I have a forked version of Steve that publishes ampere and kWh data to emoncms. (In fact there is some experimental data from a commercial OCPP1.6J EVSE at http://data.openevse.com/emoncms/vis/auto?feedid=12001 and http://data.openevse.com/emoncms/vis/auto?feedid=12043 .)  Fork is at https://github.com/chuck-h/steve

glynhudson, 2019-01-21:
As of July 2019 it's mandatory for all charging points installed under the OLEV grant in the UK to have the following smart features. This makes implementing OCPP V1.6+ over https a top priority for us. The OLEV grant is up to £500 towards the cost of the EVSE + installation, therefore is very significant. 

[image]

@jeremypoulter

DrFrankReade, 2019-01-21:
That's an amazing subsidy - I worry how the "or an equivalent" part will be interpreted, giving $$ to a company to support a closed standard, the peril being that if said provider goes out of business, the hardware is orphaned, but I digress. 

So what happens now? Does a Raspberry Pi Zero piggyback on a new OpenEVSE board with a beefy 5V buck converter and only a 12V power supply? Or the other way around? A 5V board and PSU with a little 12V boost converter to run the pilot and close 12V relays?  The price point on the dual voltage supplies is hard to swallow, and I can't find much that's got a heavy 5V rail, it's usually just the 12V that has the amperage. In looking at all the claptrap in my charger at home, an OpenEVSE, an LCD/RTC board, an ESP, a 12>5V external buck converter, and at one point an external level shifter, it's a little busy in there. Plus of course the current transformers and a contactor.  More digressing. 

Also  IEEE 1901, but at this point I need to read up a lot more before I put in any more of my two cents. I do know that ESPs are getting easier to use every day, but being able to put an image on an SD card and pop it in is pretty damn sweet rather than dragging a laptop out to an EVSE, plugging in a serial cord, pressing the reset and upload, or an OTA update that always risks stranding the device. 

A pi / SBC will allow for the giant bloated style of web interface that people have come to love, and they're so damn stinking cheap, not to mention having all of the foreseeable horsepower to do encryption, logging, and so on. 

¯\_(ツ)_/¯

jeremypoulter, 2019-02-18:
I have been looking at this with a view to (first) integrating in to the [Node.JS](https://github.com/jeremypoulter/openevse_wifi_server). My first thoughts were to do a small bridge with a suitably mature library that could run on the same device that the openevse_wifi_server is running on. However after looking at some simulators ([[1]](https://github.com/RWTH-i5-IDSG/steve/issues/53), [[2]](https://github.com/JavaIsJavaScript/OCPP-2.0-CP-Simulator), [[3]](https://github.com/JavaIsJavaScript/OCPP-J-CP-Simulator)) the actual protocol bit looks fairly simple.

What looks much more complex is the actual integration., What does it actually mean to add OCPP support to OpevEVSE, as in what is the mapping of the OCPP stuff to OpenEVSE stuff?

It would be helpful if someone with one domain knowledge could provide some guidance.

I presume the initial goal is to implement 1.6 for the OLEV grant?

SteVe has already been mentioned her, but are there any other OCPP servers that we should be testing compatibility with? 

Also, re the OLEV grant requirements, I think a case can be made that we already meet those requirements. OCPP is just a recommendation:
> ..., or an equivalent
gives you the opportunity to use something else.
 
The cyber security bit is more of a problem, the current ESP8266 version is not so great so that is more the failing point for the OLEV grant IMHO. Also on that point:
> ... any communications are exchanged in a secure manner ...
I assume they actually just mean over the network and it would not be expected that the serial comms also be secure? This also kind of implies that unencrypted comms are not allowed? Kind of makes having a local web UI a a bit challenging. So the more important question is, what do you have to do/show/present to prove your device meets those 'requirements'?

TechOverflow, 2019-02-20:
The 1.6 specification directly says that a VPN tunnel to the backend is secure enough (network level security). This would only require a router between the device and the server that is capable of making a (secure) VPN tunnel to the server and routes all traffic. I'm surprised they require it considering there's no harm that can be done to a homeowner's wallbox. There is no private data in the messages, there is no way to "steal a charge" without physical access. Even with physical access, the damage would be miniscule and not worth the effort to spoof a session. Afaik, ocpp2.0 may enable communication of tariff info, in the distant future there may be ways to send payment info, although I have my doubts. Either way, the applicant can (and should) reference the 1.6 standard's security recommendations.

I can make a logic overview explaining the software to hardware implementation if there's interest.

wbitholedotbe, 2019-02-20:
It is not about stealing or hijacking the data. It is about Tampering the
data. Changing values... If you want to submit your charge session and kwh
from your taxes this is mandatory.

DrFrankReade, 2019-02-20:
I'd argue it's about protecting the grid from attack  in the future -
Imagine on a hot sweltering day when normally all EV charging is postponed,
a bad actor instantly activates as many units at full current as possible,
or changing the pilot signal to full amperage regardless of the EVSE
physical capacity.  This is a requirement to take very seriously.

beaylott, 2019-03-10:
As part of our work with @glynhudson on a new project we will be implementing an OCPP server on a pi in order to translate requests from an upstream control server using a different protocol. This is more like a home energy management system topology (as it is with the emonpi). The standard doesnt actually seem very well suited to domestic EVSE context which makes the UK govs decision a bit weird. Is anyone aware of a lightweight implementation of the server which might be appropriate for this context?

jeremypoulter, 2019-03-10:
I agree about home use, clearly geared to large commercial installs, but they did add the nice get out clause 'or an equivalent' probably for that very reason. 

So far the only server I have been a) able to get the source and b) get working is [SteVE](https://github.com/RWTH-i5-IDSG/steve) but have not tried running it on a Pi yet.

As an FYI I am currently using https://github.com/elifTech/cpd-ocpp which is a Node.JS library I found. Has both the client and server sides for the comms layer (as do a lot of the libraries) so if you are doing a bridge that may be a good candidate.

Also I have started a [Wiki page](https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/wiki/OCPP] which I will update with more links and info from my discoveries. This thread has loads of useful info in but is kind of hard to find it.

DrFrankReade, 2019-03-11:
I have seen this "Steve" but I'm not up to date with my Java, but it definitely looks to be the most fleshed out with a UI and everything. I've really been angling to see an OCPP server implemented as a plugin for HomeAssistant (the open source home automation platform that runs great on a Pi, and if you haven't seen or used it, you're missing out) and it integrates with pretty much everything. Not even kidding about this. EVERYTHING. The plugins just run as extra servers on different ports, it's a pretty slick system. As far as Homeassistant though the integration for OpenEVSE is busted at the moment, but I was working with another guy and we got a nice little script that would talk JSON to an OpenEVSE. But I digress. Or do I? It would be better if everybody just talked OCPP and every charger did NOT need it's own special code, which is the whole point of OCPP... And homeassistant (with a Steve-like interface) makes a great user-facing experience for home use, and the opportunities for utility automation is virtually limitless.  Just saying :-)

beaylott, 2019-03-21:
I think trying to get something which will work for HA is a good idea. But as it needs to run a server does it need to be an 'add-on' for HassOS? It doesnt seem to fit the definition of a platform/component...I am not totally au fait with the HA taxonomy which also seems to be changing quite a bit ...

DrFrankReade, 2019-03-21:
Hey Ben,

Yeah you're right, and I'm a dummy for not being more careful with my language. It would almost certainly be an add-on and not a component. 

Right now, there are quite a few extra servers that run as add-ons in Hass.io - Everything from PLEX on the really heavy end down to several different MQTT brokers, DHCP servers, SQL database servers like MariaDB, FTP, Pi-Hole, and so on.  An OCPP server sounds like it would fit in that category.

Risking going off topic, is worry about wear on the SD card from excessive logging. I've already killed one, but luckily HASS.IO lets you use an external SQL database, so setting that up is my current project. I'm sure stEVe can be configured to do that.

gsameer1988, 2019-05-23:
Hello Team.

Currently we have implemented the server/CMS as per OCPP V 1.6 protocol, already tested with existing EVSE manufacturer (viz. Delta, Execom) and other OCPP V 1.6 compatible EV chargers.  We are in the stage of in-house hardware (GSM Modem with MOD Bus RS 485 interface) design and development. As I already mentioned that we had already implemented the OCPP V1.6 compatible server/cloud based back end platform but we are unable to make a hardware platform compatible to OCPP V1.6. To make this happen, I really appreciate if you can support me with supporting and integrable OCPP libraries for GSM/GPRS modem or if you can connect me with right partner who can develop the same complete hardware based solution.    

Hardware Specification: 
1. 8 Bit Microcontroller Unit: To control the inout peripheral activity 
2. RS 485 Mod Bus Interface:  To collect data from another MOD Bus Device 
3. GSM/GPRS Module: To send the collected data to OCPP V1.6 compatible server/cloud

Hardware Application: 

Collecting the data from Mod Bus enabled Slave device and sending to the OCPP V1.6 compatible server/cloud using GSM communication.

Note: Current hardware is working perfectly with HTTP protocol but we don’t have resources as such to integrate the same with OCCP V1..6 Protocol.

We are ready to buy OCPP license in subscription basis or any other associated commercial call in this direction. If the subject is under your interest, I can share the detailed product development scope for your reference. 
 
Appreciate your quick acknowledgement. For any further clarification please directly contact undersigned.

Thanks & Regard
Sameer Garg
Sr. Manager, Product Division
M: (+91) 7888098455 E: sameer.garg@magentapower.in
P: 1800-300-29-444 (toll free)
A: Sector-14, Kopar Khairane, Navi Mumbai
W: www.magentapower.in

landswellsong, 2019-06-25:
Hello everyone,
I'm considering contributing a 1.6 support, what's the bottom line about the hardware here? Esp32/Esp8266/Linux? I've read on #182 about OpenEVSE running on ESP32, do you mean the EVSE controller itself and thus a suggest a merger of both EVSE and WiFi chips eliminating RAPI?

jeremypoulter, 2019-06-25:
Those discussions are all around the 'WiFi' part, the OpenEVSE controller as staying as a separate MCU and using RAPI to communicate. 

We are currently looking at different options for replacing the ESP8266, both ESP32 and a SBC (Raspberry Pi/Orange Pi with node.js). The Node.JS version partly has OCPP support in that it has a library and make some OCPP calls but what we are lacking is some clear mapping of how OCPP fits in to the home environment. 

On the ESP32 side I am currently looking at using Mongoose as a replacement to the ESPAsyncWebserver that had a blocking bug, that being said there is now apparently a fix for that said bug that I need to try out.

landswellsong, 2019-06-25:
Right, then I'm at loss with this message by @chris1howell :
>Adafruit has releases a clone of the Huzzah ESP8266, the Huzzah ESP32 is a plug in replacement. OpenEVSE is now working on a few ESP32 boards in a couple form factors. 
If there's no ESP32 port of the WiFi and it's not about OpenEVSE itself, then what is working on ESP32? Sorry if I'm being stupid.

At any rate, when I get some free time I'll try to tinker with a native OCPP implementation and report if I have any success.

beaylott, 2019-07-11:
We have been revisiting the issue of OCPP. It is not a great fit for the case where you have a home energy management system (or generally anything in between you and the control server). Having said that, it is still compatible with OCPP to do this. But I think having OCPP support on the emonEVSE will be important going forward as there is building momentum behind it. For our project now we will just be using the existing RAPI HTTP/MQTT interface (or a new scheduling API TBD). But we have a few other projects on the drawing board which make use of OCPP. We would be happy to subsidise the  development of this capability in the context of these upcoming projects
.

justoke, 2019-07-27:
@glynhudson What is the current status? 

I noted some of the discussion around suitable hardware for what is the OCPP bridge device. If using the ESP range, in particular then there would be no need to devise hardware connections to the OpenEVSE ESP8266 - the ESP-Now wifi stack could be used - this is simply peer-to-peer data exchange over wifi. This would allow the bridge device to be installed on site in its own enclosure. And it can then be paired to the chargers.

This way a simple layer can be added to the configuration to allow one or more charge points to participate in the OCPP party. The ESP32 would be a viable solution here as it would be more than capable of handling the OCPP implementation and a single device could manage the chargers it is linked to at a specific site.

The bridge can marshall MQTT RAPI to the charger in say a OCPP site mode or if this is not needed the OpenEVSE charger can operate in standalone mode. As mentioned elsewhere in this thread, the OpenEVSE already implements some aspects of the OCPP already and it is not absolutely necessary to implement this for a home charge point.

glynhudson, 2019-08-01:
I'm still very keen for OCPP support, we don't yet have a working implementation (at least not one that I've tested). However, progress is being made which will help facilitate. 

My preference would be to implement OCPP on the ESP32 (if possible). Migration from ESP8266 > ESP32 for OpenEVSE WiFi Gateway is in its final stages. ESP32 and Mongoose will enable secure connections which OCPP requires: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/182

Also @lincomatic has added OCPP hooks in RAPI 5.0.0 OpenEVSE controller firmware https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/237

If anyone has any experience with OCPP help with implementation and testing would be appreciated.

landswellsong, 2019-08-01:
@glynhudson @justoke @beaylott I'm currently slowly trying to implement it as a hobby project, that being said I'm currently locked on the hardware part a bit, but I suggest will be done by end August early September if nobody else does it faster in the meantime.

landswellsong, 2019-08-01:
@glynhudson also I don't think Mongoose is a blocker for OCPP. OpenEVSE is in the websocket client role for the OCPP setup, I don't think it uses any web-server functionality on our side and there's a readily available lib for WS: https://github.com/Links2004/arduinoWebSockets

kernelkraut, 2019-08-06:
@landswellsong We are also trying to create a broker between openevse and ocpp server (SteVe). Is there anything I can to to help? I also have spare units here to test with.

landswellsong, 2019-08-06:
@kernelkraut yes, OpenEVSE wifi and SteVe. I'm just overloaded elsewhere so it's going very poorly.

[28 more replies]