OpenEVSE/ESP8266_WiFi_v2.x #116: OpenEVSE WiFi gateway online simulator

glynhudson, 2017-11-02:
It would be great if prospective users could browse the openevse web interface via a demo website before purchasing e.g just like http://iotawatt.com hosts a demo of the IotaWatt ESP8266 interface. We would need to simulate the EVSE controllers RAPI responses. We could also use the simulator to ealry test how the interface looks on different mobile devices etc. 

As long as the interface 'just loads' and the sensor values are populated with dummy values this will be fine. Maybe an emulator is not needed at all, we could just hardcode these values. The propose of the demo would be to give the users a preview of what the interface will look like rather than try and simulate every single feature of the openevese. 

Does this make sense? It would be easy to spend a considerable amount of time building an advanced emulator when when instead hardcoading some values would have almost the same effect.

Any thoughts @chris1howell ?

jeremypoulter, 2018-01-26:
Take a look at http://OpenEVSE.bigjungle.net/ work in progress, the WebSockets are proving to be a bit challenging to get to run in Apache the same as on the ESP, but getting there

glynhudson, 2018-01-27:
Wow, that's fantastic! Nice work 👍 

I'll reply properly when I've had a chance to fully review. But just wanted to say how impressed I am. Don't worry about getting all the features to work. IMO how it works currently is all that's needed for users to get a flavour of what the ESP openevse can offer. 

What's with `wibble_ext`?! Is this an actual WiFi network somwhere or a place holder?!

What do you think @chris1howell?

chris1howell, 2018-01-27:
The simulator looks great...

I am actually really excited about it. I have been playing with an Orange pi Zero http://www.orangepi.org/orangepizero/ It is just a little bit more expensive than the Huzzah but is WAY more powerful. I really want to start pulling data from all the vehicle APIs and push it to OpenEVSE and EmonCMS. The ESP8266 can pull from Tesla and likely BMW but I am not sure about others.

The interface may actually be very useful on other Linux hardware platforms.

jeremypoulter, 2018-01-27:
The data is a snapshot of real(ish) data from my setup. Wibble_ext is (one
of) my WiFi networks.

Figured out the WebSockets issue, bit of a pain to setup, have to alter the
server config rather than just a .htaccess file.

If you create a repo I can upload. This initial version and install
instructions.

Jeremy

jeremypoulter, 2018-01-27:
This is potentially possible with what I have. Especially with a few
changes.

That being said would it be better to integrate directly into EmonHub for
the use case you mention?

glynhudson, 2018-01-28:
>If you create a repo I can upload. This initial version and install instructions.

@chris1howell would you be able to create a repo on github.com/OpenEVSE for this? And give @jeremypoulter and myself access. 

Any ideas on what we should call it? `openevse-wifi-gateway-simulator` or `openevse-wifi-gateway-demo`?

jeremypoulter, 2018-01-30:
Hold off on creating a repository for now, I may have come up with a cunning plan....

jeremypoulter, 2018-02-02:
Behold the cunning plan, #146

glynhudson, 2018-02-05:
Merged, this issue is now fixed :+1:

closed 2018-02-05 by glynhudson