OpenEVSE/ESP8266_WiFi_v2.x #117: Use of deprecated RAPI commands for safety checks

glynhudson, 2017-11-03:
In RAPI V4.0.0 (open_evse V4.9.1+) the commands to enable / disable the safety checks are being deprecated and replaced by `$FF` see:

https://github.com/lincomatic/open_evse/commit/77fcecf0096bacb449ca31d32509c27dd1a70d14

```
+FF - enable/disable feature
 + $FF feature_id 0|1
 + 0|1 0=disable 1=enable
 + feature_id:
 +  D = Diode check
 +  E = command Echo
 +   use this for interactive terminal sessions with RAPI.
 +   RAPI will echo back characters as they are typed, and add a <LF> character
 +   after its replies. Valid only over a serial connection, DO NOT USE on I2C
 +  F = GFI self test
 +  G = Ground check
 +  R = stuck Relay check
 +  T = temperature monitoring
 +  V = Vent required check
 + $FF D 0 - disable diode check
 + $FF G 1 - enable ground check
```

glynhudson, 2017-12-28:
Started work on this in rapi-4.0.1-compat branch. Testing and backwards compatibility with V3 API needed.

jeremypoulter, 2018-03-04:
Completed the changes on this

glynhudson, 2018-03-07:
Fixed by https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/pull/145

closed 2018-03-07 by glynhudson