OpenEVSE/ESP8266_WiFi_v2.x #119: Time setting issues

glynhudson, 2017-11-10:
**1. The browser auto sync is always setting October instead of November**

[image]

***

**2. Setting time time manually results in the desired time getting corrupted**

E.g manually try and set the time to:

`01/02/1234 12:34`

[image]

Then click `Sync"

Time gets corrupted:

[image]

***

Tested on Chrome. Issues has been confirmed by @chris1howell and a few other users. 

On a related topic I personally see no use case where the OpenEVSE time needs to be set to a different time to the browser. To avoid confusion and simplify I would vote for removing the manual time setting function and just keep the auto browser sync time option. What do you think chris1howell?

chris1howell, 2017-11-10:
I concur. If manual setting is desired the time can still be set with the button.

[screenshot 2017-11-10 at 03 10 06]<https://user-images.githubusercontent.com/758844/32641233-b87b46da-c5c4-11e7-9776-d79088102be7.png>

[screenshot 2017-11-10 at 03 07 59]<https://user-images.githubusercontent.com/758844/32641203-8e8da05c-c5c4-11e7-90e3-013598c71968.png>

[screenshot 2017-11-10 at 03 08 25]<https://user-images.githubusercontent.com/758844/32641198-8b386a7c-c5c4-11e7-888e-0e75aaf75641.png>

joverbee, 2018-01-07:
in fact in GMT+1 it looks even weirder
[image]
while it really was January 7 2018 20:52

joverbee, 2018-01-10:
never mind the above comment, I am slowly learning that I don't have an RTC on my openevse and a $GT command responds with $OK 165 165 165 165 165 85 so this is unrelated to the real error
The wifi code could of course mention this as e.g. "no RTC installed" or omitting the whole time window if this is the case (better to avoid confusion)

glynhudson, 2018-01-28:
Confirmed fixed by https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/pull/132 thanks @joverbee 👍

closed 2018-01-28 by glynhudson