OpenEVSE/ESP8266_WiFi_v2.x #120: Allow some (or all) functionality while in AP mode

glynhudson, 2017-11-11:
A while back due the web server memory issues we made the change to only display the list of nearby networks to connect to when in AP mode. It's become apparent in real-world testing that being able to fully control the unit while in AP mode would be very useful 

E.g. charger is installed a remote location with no nearby WiFi network available, unit is used as a mobile EVSE therefore location is always different making connecting to a local WiFi network impossible.  

After implementing PR https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/pull/87 and https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/pull/83 to embed SPIFFS into the flash am I correct in thinking we have some more memory headroom to play with now we are not mounting SPIFFS? If so would it be feasible to enable operation of all functions while in AP mode? 

If so it would probably also be a good idea to add the ability to set the  SSID and PSK to secure the AP network when installed in public locations. I have been told by EV installers that a local admin login to be able to test the functions of the charge point would be very useful.

jeremypoulter, 2017-11-11:
Actually none of that functionality is dissabled, it is just a different
page you get back when you go to http://openevse.local/. Fetching
http://openevse.local/home.htm should give you the full UI.

Jeremy

glynhudson, 2017-11-12:
Ah awesome, that's great. I'll test it and if it works we could look ad
adjusting the captive portal re-direct URL.

jeremypoulter, 2017-11-12:
I would probably just add a link at the bottom of the page to redirect to
the full UI. IMHO the WiFi only page makes setting up a lot easier for the
majority of users and the link will allow for the use case where there is
no other WiFi.

Jeremy

glynhudson, 2017-11-13:
Good idea :+1: 

I can confirm that browsing to `/home.htm` gives us the full interface and seems to work well (no stability issues). 

I agree, it would be best not to overload a user with the full interface when all they want to do is to configure WiFi. A button is a good idea. I've added it to master branch. 

Can you think of any improvements to the wording / layout:

[image]

commit a5e1ac7b (2017-11-13): add option to continue in AP mode fixes #120
closed 2017-11-13 by glynhudson