OpenEVSE/ESP8266_WiFi_v2.x #121: Main web page won't open, but RAPI page will over a VPN

DrFrankReade, 2017-11-14:
Hello,

So weird stuff is going on... When going in through a VPN (my own VPN screwed to a wall in my garage) to get to my home charger, some stuff is working but some is NOT. 

http://10.44.63.182/r?rapi= works and I can send commands, BUT
http://10.44.63.182/ gives me the spinning wheel of forever. sometimes just the Open
http://10.44.63.182/foo gives me a 404

If I open up port 80 to the outside world and access the OpenEVSE through my router and my public IP, it DOES work, but that's not cool for a number of reasons I don't need to go in to. 

Every other device on my network works great with the VPN - Security cameras, chicken coop, etc so I'm not immediately ready to blame it all on my VPN (OpenVPN)- I'm hoping that this pattern of what works and what doesn't might lead to a discovery of what's causing this. 

Any ideas on this?

glynhudson, 2017-11-15:
Thanks for reporting 

My experience accessing the unit using port forwarding from a dynamic dns RAPI commands fail. This makes all the reading on the webpage blank. I think the issues we are experiencing are related. I think the URL path is not being processed correctly.

Does your unit work ok when accessing locally? What FW version are you running?

[image]

Hopefully @jeremypoulter might be able to take a look soon.

DrFrankReade, 2017-11-15:
Hello,

So I CAN access it locally. BUT I can also access it through port forwarding,although I just tried it again and this time it didn't work... The weird part is that I can't access it over the VPN, which is for all practical purposes a local machine as far as my network is concerned. I am however coming in from a different subnet (I think?) the VPN is 10.8.0.3 and the OpenEVSE is 10.44.63.182 BUT some pages work, like the RAPI page, but everything else just doesn't. I won't rule out something in the VPN server configuration, but virtually every other device can talk just fine, and of course I can ping the OpenEVSE. Unclear, especially because that one page works. I'm going to screw about with my subnet settings and update.

chris1howell, 2017-11-15:
Andy can you try setting the MTU on your host machine to 1420? You may be exceeding the maximum size of 1500 when the encryption headers are added. If a don not fragment bit is set the packets would be discarded.

DrFrankReade, 2017-11-16:
Hello,

I'm running wifi firmware 2.4.2

From what I've read, it SOUNDS like an MTU problem - First part of the site loads, the rest doesn't, but I'm having trouble fixing it or even sniffing out the problem.

New symptom, when I enable HTTP authentication, and I go to 10.44.63.182 it asks for login, and I give it to it, and it goes back to waiting. If I give it the WRONG password, it DOES ask for it again, so we know the login is getting to the OpenEVSE. 

When I go to http://10.44.63.182/r?rapi=%24SC+40 it will ask for authentication, and then will execute the RAPI command. 

I'm also not 100% clear on setting the MTU - I know that I CAN change the MTU, and I know that the size of the packet I can ping changes accordingly, so I think that's doing what it's supposed to, and I've tried various fragmentation settings to no avail. But stupid forwarding port 80 through the router seems to work fine. At this point, I don't know what the fundamental difference is and I'm out of my own ideas.

glynhudson, 2017-11-17:
>New symptom, when I enable HTTP authentication, and I go to 10.44.63.182 it asks for login, and I give it to it, and it goes back to waiting. If I give it the WRONG password, it DOES ask for it again, so we know the login is getting to the OpenEVSE.

There is currently a known issue with HTTP auth https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/101

Sorry I'm not familiar with VPN MTU settings.

glynhudson, 2019-09-09:
This is now fixed on the ESP32 V3 version http://github.com/openevse/ESP32_WiFi_V3.x/

closed 2019-09-09 by glynhudson

soloam, 2019-10-07:
Hello, is this only solved in the ESP32? I have a system with a ESP8266, Will it solved or do I need to change to a ESP32?

Thank you

glynhudson, 2019-10-08:
This is only fixed for the ESP32 version, I would recommend upgrading. However, the ES32 version is working now but still in beta.

soloam, 2019-10-08:
I will order a ESP32 to test out and help in the Beta debug! Any recommendations on the ESP32 board?

Thank you

glynhudson, 2019-10-08:
Lots of boards are supported e.g NodeMCU, Huzzah Feather, Olimex ESP gateway. However, I prefer the ESP32 Huzzah, since it easily connects to the same UART https://www.adafruit.com/product/4172. 

You will need to compile and upload using `$ platformio run -e openevse_huzzah32 -t upload`

See `platfromio.ini` for the names of other supported boards https://github.com/OpenEVSE/ESP32_WiFi_V3.x/blob/master/platformio.ini