OpenEVSE/ESP8266_WiFi_v2.x #122: Travis CI Failing

glynhudson, 2017-11-15:
Issue: Travis CI arduino build is failing since we have switched to using the new SDK. 

Solution: I vote for removing Arduino IDE from the travis build since we cannot easily specify version numbers for SDK for libraries.

glynhudson, 2017-11-15:
I've made a start: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/c369de9e966085131a9aa7210cd65b6dbe2d7c00

SPIFFS build is no longer needed. I've also simplified b just building the `openevse` environment.

jeremypoulter, 2017-11-15:
No, no, no. Put that all back you bad person ;-)

The whole point of CI is to check all the configs. If they are truly
redundant then you should also clean them up from the source.

Also about the Arduino IDE built this still is popular with folks so we
should fix it rather than remove it.

glynhudson, 2017-11-15:
I did wonder if you would intervine! This is why I only commented out the Arduino checks.

I understand the argument for a comprehensive build test, however it's the `openevse` environment is what we really care about. I guess it would be a good idea to also test `openevse_dev`. However, I'm not sure if testing `openevse_staging` is worthwhile since it might often fail as arduino SDK and the libraries are changed. 

I don't believe the Arduino IDE test is essential, for a relatively complex application like this I think being able to specify version numbers of libraries is essential

Basically I'm keen to test things we are critical and don't worry about things that are not.

jeremypoulter, 2017-11-15:
But that is exactly the reason to have it so you know what builds and what
doesn't. You know if uplifting to a new library is just a case of
compilation and testing or if some code needs changing.

A failed build is just that. Although it is good to have everything passing
for a release, for example, it doesn't have to and it if much better to
know about the failures and make an informed decision rather than assume
everything is ok.

IMHO

glynhudson, 2017-11-16:
I've added back in to test all PlatformIO environments apart from stagin: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/97f19881444e44ad8dff9531370144dcc4abd067

IMO the reason we specify specific versions of libraries in `platformio.ini` is to put us in control over when and how things change. I'm worried testing staging environment using latest libraries could cause a straightforward clean commit/PR build to fail if the master branch of any of the libraries has an issue, this is no fault of ours. I would rather use the staging environments to test comparability with latest SDK / 3rd party libraries at my leisure when we decide that a platform update is required. Unfortunately since a Travis build is black/white (pass/fail) it might return a false negative failed build when there is nothing actually wrong with our code or the default stable environment.

closed 2017-11-24 by glynhudson