OpenEVSE/ESP8266_WiFi_v2.x #127: Add option to update multiple Emoncms Servers under Energy Monitoring

cmgj74, 2017-12-29:
This would be nice for testing on another server (running at home compared to using data.openevse.com/emoncms, emoncms.org, emonpi/emoncms )

It also could be used to cut over to another server, or test other versions of the emoncms server.

Thanks!

glynhudson, 2018-01-01:
I agree, it could be useful. However since this esp8266 is resource limited
IMO I think posting to multiple emoncms servers is best achieved by posting
to an mqtt server then using Node-RED or mqttwarn to forward the data onto
as many services as you wish.

- sent from my mobile device

cmgj74, 2018-01-02:
Ahh, thank you...I have setup Node-RED and used MQTT and now have data going to 2 different emoncms servers.  Thanks a lot! I see more uses for all of this now too :)

closed 2018-01-02 by cmgj74