OpenEVSE/ESP8266_WiFi_v2.x #130: showing 3 phase session charge energy

joverbee, 2018-01-07:
As it seems to be impossible without significant hw changes to show the correct charge energy per session for 3 phase systems, I slightly modified the code to get this:
[image]

I have a portable charging station, so depending on where I am, I know whether to look at the single phase energy or when to look at the 3 phase session energy (at home). It's not perfect (auto detect would be better, or a gui switch between 1p/3p, but this does the trick well enough for me)

if interested, see modif in home.htm in 
[https://github.com/joverbee/ESP8266_WiFi_v2.x/commit/a166e6fc9455ce00d0c1252413d0ff9dac80ad48](url)

chris1howell, 2018-01-07:
You could use the L1/L2 detection mechinism to identify 1 or 3 phase. If
you put the 1st AC_Test line on P1 and the second on P3 then if only P1 is
hot it is single phase. If P1 and P3 both read hot then it is three phase.

To calculate 3 phase power you multiply voltage by current by the square
root of 3 or 1.73. It looks like the calculation on your screenshot is off.

<https://user-images.githubusercontent.com/5755572/34653384-500d0ea8-f3eb-11e7-80d2-5e9e2fcc2a01.png>

https://github.com/joverbee/ESP8266_WiFi_v2.x/commit/
a166e6fc9455ce00d0c1252413d0ff9dac80ad48 <http://url>

<https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/issues/130>, or mute the
thread

joverbee, 2018-01-08:
hmm, I see
Indeed the factor 3 I used might be wrong even though I thought that the sqrt(3) is already taken into account by the fact that we stil assume 230V while the potential between the lines is now 400V.

From your comment I deduce that this would be more something to change in the openevse firmware so that also on the lcd display it appears correct. The wifi firmware will keep showing whathever energy is relevant. Will check if my setup has this possibility of knowing which phases are 'hot'.

closed 2018-01-13 by joverbee

glynhudson, 2018-03-13:
Hi @chris1howell 

I believe @joverbee is correct. * √3 is required to convert line-to-line voltage (usually around 400V) to the line-to-neutral voltage (usually around 230V). Total power is then calculated by multiplying the current in each phase by the line-to-line voltage (230V). Since we assume L2 to be 230V on the OpenEVSE it's a simple task of multiplying the energy by 3.

```
- P₃ₚₕ = Vʟɴ * I * 3
- Vʟɴ = 230V (line to neutral voltage)
- Vʟʟ= Vʟɴ * √3 = 230 * 1.75 = 400V
```
I've opened a PR on the open_evse FW to add the option to enable 3phase energy calculation: https://github.com/lincomatic/open_evse/pull/84.

While it's possible to detect if the EVSE is connected to a 3-phase supply using the L1/L2 auto detect tick you mention above, I don't think it's possible (without extra hardware) to determine if the EV is drawing on single or 3-phase. e.g. Tesla / Nissan LEAF using the same EVSE. To do this will will need an additional power CT on L2. Putting L1,L2 and L3  through the same power CT will result in zero when charging on 3-phase.