OpenEVSE/ESP8266_WiFi_v2.x #133: Max Utility Power Mode

glynhudson, 2018-01-11:
Add feature to adjust charge rate to supply maximum power up to a maximum grid import threshold. I.e reverse of solar PV divert. 

Enter total grid import MQTT feed (including EVSE power) and maximum utility supply threshold. 

Useful for users with a limited gird supply who want to charge their EV at maximum rate, charge rate will be automatically reduced to keep grid import below max utility supply threshold e.g. if another appliance is switched on charge rate will be reduced. 

If possible this feature should be independent to solar PV divert since it could be possible for a user to want to use solar PV divert e.g. "Eco Mode" and "Max Utility Power Mode". However given the memory constraints of the ESP it may be easier to implement max power mode using the same solar PV divert functions. It would not be too bad trade off to make the user choose between Eco Mode or Max Power mode e.g not both at the same time. 

*Note: recomend setting max utility grid import threshold below actual threshold to give some headroom to avoid overloading supply.*

perefeliu, 2018-04-11:
Funny! I bought recently an Emonpi and an OpenEVSE with the Wifi gateway to implement this function. By the way, congratulations to the team involved on both OpenEVSE and OpenEnergyMonitor designs, it has been very straightforward to implement. 

I agree that a user could want to use both solar PV divert and Max Utility Power Mode at the same time (it is wiser to reduce power in case of solar generation AND excessive grid import).

If this function is  implemented as this, there will be three sub-modes inside the Eco Mode: PV divert, Grid (+I/-E) and Max Utility Power Mode (with an text field for the maximum grid import).

It could be presented as folows:
Text field 1: for SolarPV-gen topic (feed should represent generated PV power).
Text field 2: for Max Utility Power topic (feed should represent the grid import) & Text-field 3: user should introduce utility power threshold (greater than 6A*240V).
Text field 4: for Grid (+I/-E) topic (feed should also represent the grid import). 

Text field 1, 2 and 3 are not exclusive and they can be edited simultaneously.
Text field 4 edition should disable the text fields 1, 2 and 3.

In my case, I ended up creating a unique MQTT feed that represents the power setpoint that I want my EV to follow, which is different deppending on what I want to achieve. I use the PV divert option so the feed represents directly the power setpoint for the EV. The equations that generate my setpoint feed are given as follows for each mode:
PV divert: Ppv
PV divert + Max utility import: Ppv - onlyPositive(Pimp - Pmax + Ppv - Pev)
Max utility import: Pmax - Pimp + Pev
Grid (+I/-E): Pev - Pimp

Where, 
Ppv is the power PV generated.
Pimp is the power imported from the grid.
Pmax is the maximum admisible power from the grid.
Pev is the EV charging power.

In this case, it is not the Wifi Getaway that choses the mode, but rather the MQTT feeder. This has the advantage of much more simple UI and code for the wifi getaway but it requires a little bit of work for defining the feeds ( maybe it shouldn't be expected from users to be that technical).

zymurgic, 2021-09-14:
Interesting proposed legislation in the UK, we may need to adjust the maximum grid import for EV charging down to cater for local distribution network operator grid loading, exact details of how to determine if load shedding is required at any point in time, and whether it'll be mains frequency analysis as used for other industrial load shedding or via some internet-reachable API endpoint.
Also: random delay requirement, relating to peak/off-peak timed charging, presumably to prevent massive surges as everyone's EV starts charging at the end of a mandated peak time.
https://assets.publishing.service.gov.uk/government/uploads/system/uploads/attachment_data/file/1015285/electric-vehicle-smart-charging-government-response.pdf