OpenEVSE/ESP8266_WiFi_v2.x #137: Hardware reset

glynhudson, 2018-01-25:
Add the ability to clear all EEPROM settings by holding down a button (GPIO0 / boot) at startup to wipe all EEPROM settings and factory reset the device. Would be a fail safe for #118 or if the http authentication password has been forgotten. 

A number of users are managing to lock themselves out of the device.

CliffStoll, 2018-04-03:
Yes please!   The inability to reset the wifi password has wasted several hours.  And it's silly to have to reprogram the system just to clear the wifi password.  Imagine if your cellphone worked that way...

chris1howell, 2018-04-03:
This was part of 1.x...

It would make more sense to go into AP mode with a long press of the
external button via RAPI. And use the Reset/hardware button for factory
reset.

while (buttonState == LOW) {
  buttonState = digitalRead(0);
  erase++;
  if (erase >= 5000) {
    ResetEEPROM();
    int erase = 0;
    WiFi.disconnect();
    Serial.print("Finished...");
    delay(2000);
    ESP.reset();
  }
}

chris1howell, 2018-04-20:
I have reprogrammed several boards this week due to forgotten passwords. Users are frustrated with the process to erase/reprogram.

glynhudson, 2018-05-01:
Noted, we'll get onto this. sorry for the hassle.

glynhudson, 2018-05-16:
Done, implemented in https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/pull/180

closed 2018-05-16 by glynhudson