OpenEVSE/ESP8266_WiFi_v2.x #140: 3-phase support?

glynhudson, 2018-01-29:
@joverbee I notice you have been using openevse with 3-phase. Ref issue #130 and your commit https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/pull/139/commits/a166e6fc9455ce00d0c1252413d0ff9dac80ad48 that was subsequenatly removed. 

It would be great if the WiFi gateway was able to support displaying 3-phase current when used with 3-phase systems. Do you think it would be feasible to add a config option to turn on 3-phase current display? What are your thoughts on this? 

How do you currently calculate and display 3-phase current on your setup?

glynhudson, 2018-01-29:
Having re-read #130, it seems the best way is to change the openevse FW. I will take a look into this 

https://github.com/lincomatic/open_evse/tree/development

closed 2018-01-29 by glynhudson

joverbee, 2018-01-30:
I think this is perfectly possible with the L1/L2 detection, but it seems indeed more logical to do it in the OpenEvse firmware where it is relatively trivial with the line voltages. However, to upload new firmware on the OpenEvse I need a programmer which I currently don't have and I risk bricking it which would put me without transport ;)

glynhudson, 2018-01-30:
Uploading FW to open_evse controller is not as scary as it sounds. It very
easy to restore old FW in the advent of disaster. If you would like to work
on this I would be happy to send you an openevse controller to test with.
Alternatively if you wanted to make some changes to the openevse FW I would
be happy to test.

This is our branch of latest openevse firmware:
https://github.com/openenergymonitor/open_evse

I've been running V4.11.0 for many months now without issue on my open_evse

beikeland, 2018-07-09:
Its easy enough to use a spare arduino (or even ESP8266 with level shifting) as an programmer. Once you have an in circuit programmer its virtually impossible to break the OpenEvse controller to the point where you can't recover by flashing the unmodified firmware.

In Arduino IDE go Files -> Examples -> 11. ArduinoISP that will turn an Arduino into a programmer.

In regards to L1/L2 detection, how would it allow us to detect 3-phase supply and, maybe more importantly, distinguish between single phase and 3-phase charging?

joverbee, 2018-07-09:
Indeed, it doesnt allow automatic detection of 1/3 phase as I wrongly assumed. I also don't see immediately an easy way to make the ac line test schematic do this without changing the schematic.