OpenEVSE/ESP8266_WiFi_v2.x #141: RFC: option to exclude openevse current from divert mode calcs

sandeen, 2018-01-30:
Started playing with divert mode today now that I have an mqtt feed set up for my net grid usage, and it's pretty cool, but I have a bit of a weird situation: my openevse branch circuit is not actually part of my main grid CT measurements.  So to fake it out, I grab amps off the openevse mqtt feed, and add it to the mains.  This is ... suboptimal.  And, I think the fairly slow feedback loop leads to divert mode bouncing around.

An option to exclude openevse current from the divert mode calcs, i.e. "my grid feed measurements do not measure the EVSE" would be useful - it's just a small change in divert.cpp where 
```
      // Subtract the current charge the EV is using from the Grid IE
      if(0 == rapiSender.sendCmd(F("$GG"))) {
        int milliAmps = String(rapiSender.getToken(1)).toInt();
        double amps = (double)milliAmps / 1000.0;
        DBUGVAR(amps);
        Igrid_ie -= amps;
        DBUGVAR(Igrid_ie);
      }
```
could be made conditional.  Creating the toggle in the UI and plumbing that through, however, looks daunting to me.

I think this could be useful in general - at least in my area, the utility offers some EV rates only with a completely _separate_ EV service and meter, so a premise energy monitor which doesn't include the EVSE may not be _that_ unusual.

Thanks,
-Eric

jeremypoulter, 2018-01-30:
I am not totally sure I understand your setup, would it be possible to draw a diagram of how stuff is connected together? From my initial reading it is unclear what you are 'diverting' if the two circuits are unrelated, but I think a diagram showing all the circuits an measurement points may help.

glynhudson, 2018-01-30:
> Started playing with divert mode today now that I have an mqtt feed set up for my net grid usage, and it's pretty cool, but I have a bit of a weird situation: my openevse branch circuit is not [...]

Is it possible for you to get a MQTT feed of your solar PV generation (instead of gird usage) and use this it enter into the OpenEVSE? As you have realised grid import is supposed to be *actual* grid import (including openevse consumption)

sandeen, 2018-01-30:
This is essentially what I have, the EVSE circuit is outside the house+solar metering, but I'd still like to be able to set the EVSE to consume the net grid export from house+solar.
[image]
There is a rate plan here which requires this setup - a completely separate service + meter - for an EV utility rate.

sandeen, 2018-01-30:
The help for the divert mode says 

>  Note: It's assumed that EVSE power is included in the gird [sic] feed

and in my case that assumption is incorrect.  I can work around it by adding it in via a RAPI query but if anyone else has the setup above, it might be a useful switch to have.

sandeen, 2018-02-08:
Fine by me to close this if you don't think it'd be useful to anyone else, I can work around it.

sandeen, 2018-03-28:
I may renew this request.  I had worked around this issue myself by taking the mqtt feed from OpenEVSE and adding that into the house load, and republishing the sum.  But there seems to be a bit of feedback/hysteresis problem.

[image]

I think that the ability to exclude EVSE power consumption from the divert calcs would benefit anyone who has a submetered (or unmetered) EVSE.

glynhudson, 2019-01-21:
IMO gird reading should be grid reading, since this is a usual case where the openevse is not included in your 'grid' reading it's not a high priority to implement this feature and could cause confusion. As you have mentioned there are work around using a re-published MQTT feed with open-evse data subtracted.

closed 2019-01-21 by glynhudson