OpenEVSE/ESP8266_WiFi_v2.x #143: Remove RAPI, successful packets & Emoncms post count?

glynhudson, 2018-01-31:
I don't see the need to count RAPI, successful packets and Emoncms successful posts. No reason to waste constrained system resources if this is not necessary.

Would you be happy if they were removed @chris1howell? Can you see any unwanted side effects from doing this @jeremypoulter? 

[image]

[image]

chris1howell, 2018-02-01:
I agree, those counts were meant for debugging. My plan was to replace them with "Last sucessfull" color coded timer like on the EmonCMS.

jeremypoulter, 2018-02-06:
From a memory POV I don't think those counts take up a lot of space in the grand scheme of things, and I find the EmonESP equivalent very useful as I log them in EmonCMS:

lincomatic, 2018-02-09:
I think they should be left in. The resource usage is minimal

sandeen, 2018-02-24:
I used them in the past when I was playing with RAPI, mostly just to see if we were correctly accounting for all commands issued and received.
Just my $0.02 - keeping them around for debugging seems fine, but most users aren't going to have any use for them, so from a clutter perspective, I could see moving them under the "Advanced" toggle.

jeremypoulter, 2018-03-01:
Don't think there is anything to gain by removing these and there is a reasonable amount of support, so closing

closed 2018-03-01 by jeremypoulter

glynhudson, 2018-03-01:
Sure, fine by me. A user recently mentioned to me that he would prefer all the config parts of the interface to be hidden by default to make it easier for other family members to use, just having the daily 'session' options visible e.g start/stop/timer etc to avoid confusion and accidentally editing the config. I see his point, but I don't want to dumb down the interface too much, having the services config in a separate tab is a good separation. However, it may make the interface faster to load if we only loaded a few key elements by default. Maybe just something to keep in mind that we should try and hide all the debugging parts of the interface to be hidden under the advance tab.

lincomatic, 2018-03-02:
If it isn’t too much work, it would be nice if the advanced/debugging settings would stick.
Kind of a chore to have to activate both of them after every refresh/reload of the page.

jeremypoulter, 2018-03-04:
> If it isn’t too much work, it would be nice if the advanced/debugging settings would stick.
Kind of a chore to have to activate both of them after every refresh/reload of the page.

Done as it was annoying me too, #161. Just wish I had done it before completing #117  ;-)