OpenEVSE/ESP8266_WiFi_v2.x #149: Max Current ability to set Lower than 10A

cmgj74, 2018-02-08:
This could be helpful.  Thanks!

glynhudson, 2018-02-08:
Yes, V4.11.0 support 6A - 32A a well as other fixes. 

https://github.com/openenergymonitor/open_evse/releases 

The OpenEnergyMonitor open_evse repo is tracking the [lincomatic/development](https://github.com/lincomatic/open_evse/tree/development) controller FW branch with customisation for UK/EU e.g 32A limit and set L2 as default.

Alternatively you could compile from lincomatic dev branch for latest FW. Check the change log carefully https://github.com/lincomatic/open_evse/tree/development

cmgj74, 2018-02-08:
> open_evse/tree/development

I assume this version works in the US as well?

Thanks!

Cory

glynhudson, 2018-02-08:
Yes, you will just need to set L1 if your using L2. However you should
really be using L2 if you have an OpenEVSE! :-)

closed 2018-02-08 by glynhudson

cmgj74, 2018-02-09:
I can currently charge up to 40A....this is limited to 32A?