OpenEVSE/ESP8266_WiFi_v2.x #151: negative temperature

lincomatic, 2018-02-09:
One of my EVSEs only has a MCP9808. As such, the only valid temperature is TEMP2. Therefore, my displayed temperature is garbage:

[image]

Would it be possible to test the temperatures for negative values in sequence, and use the first positive value encountered?

glynhudson, 2018-02-12:
The issue with looking for the first non negative value is it's possible for the temperature inside the openevse to be negative in very cold environments.

Is the value -256 a preset 'dummy' value for when no sensor is connected? Could we look for this instead?

lincomatic, 2018-02-13:
It is, but unfortunately, older versions of the firmware returned zero or 230. I didn’t discover this until recently.. someone else wrote the code.
I changed it in D4.10.0

vD4.10.0 20170817 SCL
- change TMP007 temperature was showing up as 230 when not installed.
- also, temperature for other devices was 0 when not installed, but 0 is a
               valid temperature
               -> change all default temps to TEMPERATURE_NOT_INSTALLED (currently
               -2560)
               -> bump up RAPI version to 4.0.1

From: Glyn Hudson [mailto:notifications@github.com]
Sent: Monday, February 12, 2018 3:47 PM
To: OpenEVSE/ESP8266_WiFi_v2.x <ESP8266_WiFi_v2.x@noreply.github.com>
Cc: lincomatic <lincomatic@hotmail.com>; Author <author@noreply.github.com>
Subject: Re: [OpenEVSE/ESP8266_WiFi_v2.x] negative temperature (#151)

The issue with looking for the first non negative value is it's possible for the temperature inside the openevse to be negative in very cold environments.

Is the value -256 a preset 'dummy' value for when no sensor is connected? Could we look for this instead?