OpenEVSE/ESP8266_WiFi_v2.x #152: Add Tesla Client

chris1howell, 2018-02-10:
Add a Tesla client to pull Battery and temperature information.

Create Tab for vehicles

Add UI for Tesla credentials
Save Tesla Credentials to EEPROM (Alternate use credentials to get a token and discard, user would have to enter credentials periodically)
Add UI to pick vehicle (required if Tesla account has more than 1 vehicle)
Add RAPI to display battery stats to bottom line of LCD while vehicle is connected.
Add UI to allow Charge Vehicle to xxx% or xxx Miles.

`// TESLA Vehicle JSON API by OpenEVSE
// API Documentation https://timdorr.docs.apiary.io
//=====================================================

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

//SET Network and Tesla Account 
//=====================================================

const char* ssid = "";
const char* password = "";
const char* tesla_user = "";
const char* tesla_pass = "";

// Use web browser to view and copy
// https://www.grc.com/fingerprints.htm 
// SHA1 fingerprint of the certificate
//=====================================================

const char* fingerprint = "11:AB:EF:76:55:1F:EC:DA:31:51:61:26:97:68:FD:13:9A:04:D6:B6";
const char* tesla_host = "owner-api.teslamotors.com";
const int httpsPort = 443;
const char* TESLA_CLIENT_ID = "81527cff06843c8634fdc09e8ac0abefb46ac849f38fe1e431c2ef2106796384";
const char* TESLA_CLIENT_SECRET = "c7257eb71a564034f9419ee651c7d0e5f7aa6bfbd18bafb5c5c033b093bb2fa3";

//Static Vehicle selector until UI for choice is implemented
int vehicle = 0;

//Strings populated once authenticated with TESLA servers
String tesla_access_token = "";
String tesla_vehicle_id = "";

WiFiClientSecure client;

void setup() {
  Serial.begin(115200);
  //Serial.println();
  //Serial.print("connecting to ");
  //Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    //Serial.print(".");
  }
  //Serial.println("");
  //Serial.println("WiFi connected");
  //Serial.println("IP address: ");
  //Serial.println(WiFi.localIP());

  delay(1000);
  //Serial.print("connecting to ");
  //Serial.println(tesla_host);
  
  if (!client.connect(tesla_host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

[... 203 lines cut ...]
      delay(100);
    }
    for (int i=0; i <= 20; i++){
      Serial.print("$FP 0 1 ");
      Serial.print("Done in - ");
      Serial.print(response_time_to_full_charge);
      Serial.println(" Min ");
      delay(100);
    }
    for (int i=0; i <= 20; i++){
      Serial.print("$FP 0 1 ");
      Serial.print("Voltage - ");
      Serial.print(response_charger_voltage);
      Serial.println(" VAC ");
      delay(100);
    }
    
    
    break;
  }
}

  delay(2000);
}
`

jeremypoulter, 2018-02-10:
Is there a way to test this for those of us who don't have a Tesla?

chris1howell, 2018-02-10:
Jeremy, contact me directly and I will supply login details. I also have a python script in addition to the ESP8266 code I posted if you want to see it...

sandeen, 2018-02-24:
It'd be pretty cool to have car stats in general; Nissan also has an API (albeit woefully slow).
I've wondered about using an ESP32 w/ bluetooth and a BT OBDII dongle to gather data directly from the car, as well.  </dream>  I guess that's a lot of scope creep.  :)

jeremypoulter, 2018-03-07:
I wonder if it would make more sense to have the ability to pipe the car stats in via MQTT, with something with a bit more powerful interfacing with the car? While It doesn't look to challenging to implement Tesla, especially if a lot of the initial setup is done in the Web UI, I assume we will want to add support for additional cars as @sandeen eludes to. 

I guess it depends how standalone you want the system to be.

glynhudson, 2018-03-07:
Yes, having an MQTT topic with a defined structure that the openevse wifi can subscribe to for EV specific information is a good solution I think. Since over time we will no doubt want to support an ever increasing number of EV's. It seems limited resources of the ESP does not really make it suitable for handling EV API calls. 

I agree, this would be trading off the simplicity of a standalone unit for more power elsewhere. However, it does seem that most users who would be interested in doing this are the same sort of users who are interested in diverting solar PV and other home automation / monitoring which does require an additional hub e.g. Raspberry Pi / emonPi / Synology NAS etc. . If we had a python service to post EV status to MQTT I could include this pre-installed on emonPi.

beikeland, 2018-07-08:
Ah, there is work in progress! Its possible to do some testing against https://private-anon-9531e42980-timdorr.apiary-mock.com one downside seems to be the json is multi line from the mock server.

chris1howell, 2018-07-09:
@beikeland The Tesla client above is fully working on the ESP8266 with the production Tesla Servers. The developers working on this are not sure this process is best handled by the client with its limited resources especially when you start adding other Manufacture/Vehicle clients. 

We are currently investigating handling this on a server and sending the data to EmonCMS. Then having the WiFi client pull down the current values through an emoncms/input/get call.

This eliminates the need for an additional MQTT device.

beikeland, 2018-07-10:
I was perhaps unclear on why I used the term work in progress; integration to the web configuration is lacking; which is where I started browsing issues and PR's to see if I could find more details on how the html/js/c is tied together.

I would rather not have to ask someone to give me their credentials to hard code; and only ever let the code have access to the token to eliminate the possibility of starting the vehicle if exploited.

But I see there are also issues on changing the configuration so, I guess I'll teach my friend to flash this thing when he needs to update tokens. Use case is calling the `command/charge_port_door_open` api endpoint when a photo interrupter detects the charge plug has been removed from the holder.

glynhudson, 2020-07-28:
Implemented in ESP32 version https://github.com/OpenEVSE/ESP32_WiFi_V3.x/pull/97

closed 2020-07-28 by glynhudson