OpenEVSE/ESP8266_WiFi_v2.x #153: Simulator not working on Safari

glynhudson, 2018-02-12:
I've setup a simulated version of openevse using the nodejs app served via apache at: https://openevse.openenergymonitor.org

It works fine on Chrome, Firefox and Android (all the platforms I have access to). However a user (@davehun) has reported page not loading on Safari on mac and ios. The page hangs at "loading please wait" with the following error in dev console:

`Exception with thrown value: SyntaxError (DOM Exception 12): The string did not match the expected pattern.`

A stackoverflow suggested that the issue could happen on older version of jqueary, I tested updating to latest version of jqueary v3.3.11 but the issue had persisted. 

Do you have access to an ios device? Are you about to recreate this issue? 

I'm currency running the `secure-sockets` branch since I needed to enable secure web sockets since the page is served over https.

jeremypoulter, 2018-02-13:
Does the ESP8266 version work? If so probably something to do with running over HTTPS. I do have a Mac/iOS so will check later today.

glynhudson, 2018-02-13:
Yes, the version directly from the ESP works on IOS. Or at least it did
last time I tested it. I don't own an ios device

- sent from my mobile device

davehun, 2018-02-13:
yes current version is fine.
Ill try and break out the debugger on it at some point. There is a JS exception going off but its being caught and swallowed.

davehun, 2018-02-14:
ok so a log of debugging.
If you make the rapi end point fail (e.g. set the port wrong) then everything loads fine.
If the + "?json=1&rapi=" + encodeURI(c), function(c) { works then on safari the home.js never loads successfully.

davehun, 2018-02-14:
essentially initialised never gets set to true so
  <div class=container data-bind="visible: initialised" style="display: none">
is never displayed.
you can remove the bind and style:
  <div class=container >
and the simulator runs ok

glynhudson, 2018-02-20:
I should add for the aid of debugging that the current simulator version running on the server is using secure web sockets as commited to the secure-sockets branch:

https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/commit/cf9db580a91e51b6ad72c40f83a825eb0e9648fd

To merge the secure sockets branch into master we will need to add some check to see if the input is via http or https and use secure/un-secure web sockets accordingly. 

e.g. 

```
var loc = window.location, new_uri;
if (loc.protocol === "https:") {
    new_uri = "wss:";
} else {
    new_uri = "ws:";
}
new_uri += "//" + loc.host;
new_uri += loc.pathname + "/to/ws";
```

https://stackoverflow.com/questions/10406930/how-to-construct-a-websocket-uri-relative-to-the-page-uri

closed 2018-02-26 by glynhudson